loicland/superpoint_graph #2: If the intensity value in SEMANTIC3D dataset is used or not?

LZDSJTU, 2018-01-27:
First I want to ask that if you use the intensity value in your code as I do not find you use it.

Secondly, I want to ask you if you know the meaning of the negative intensity value in semantic3d dataset. As far as I know, the intensity of LIDAR is usually positive as I only use the Velodye LIDAR.

Thank you very much!

loicland, 2018-01-27:
We did not use the intensity in this data set as it seemed too noisy.

I am not aware of the details of the preprocessing applied to the intensity value by Hackel et al to obtain negative value neither.

LZDSJTU, 2018-01-28:
Thanks! I have another question. Can I just use xyz without rgb to generate semantic labels using your code with little modification?

loicland, 2018-01-28:
Yes it should work if your classes are distinguishable purely from their geometry. No change of code should be necessary, change pc_attribs and ptn_nfeat_stn.

closed 2018-03-06 by loicland

otaj, 2019-02-02:
Hi, sorry about necrobumping @LZDSJTU, however I emailed Hackel and it seems, the intensity was originally captured to 12bit signed int, therefore in the range of -2048, 2047. You can of course rescale it however you seem fit