loicland/superpoint_graph #3: Loading of trained network

JD-ETH, 2018-01-31:
I am trying to load the trained network as in instruction, but can't seem to unzip the tar file from the link. 

I was running following command:

`tar -xvzf  model_sema3d_trainval.pth.tar
`
And receive the following error message:

> gzip: stdin: not in gzip format
> tar: Child returned status 1
> tar: Error is not recoverable: exiting now

Any idea why this is happening?

JD-ETH, 2018-02-02:
@mys007  Thanks for the comment. I am not familiar with pytorch :P 
This issue can be closed now. I found another small issue that the cloud_precomputation needs to be manually done before running the saved model:

`python learning/sema3d_dataset.py --SEMA3D_PATH=$SEMA3D_DIR`

Otherwise the parsed/ folder still shows empty.

closed 2018-02-02 by JD-ETH

mys007, 2018-02-02:
Yes, we mention this point in the README in the Training section, but it is indeed necessary for testing as well, thanks.