loicland/superpoint_graph #6: Apply spg on other dataset

laoreja, 2018-02-08:
Hi,
thank you for your excellent work!
I'm wondering what's the required dataset format if I want to apply your network on other point cloud dataset?
And if possible, is there any instructions on which part of the code I should modify to make SPG work on other dataset?
Thanks!

loicland, 2018-02-08:
Hi,

For the partition part, you only need to add to ```/partition/provider.py``` a function which takes the path to a file in your given format and outputs ```xyz, rgb, label``` (note that one such method is already implemented for ply files). Then copy the ```partition_semantic3d.py``` script and change L83 so it calls your function (you will need to adapt the folder browsing structure to how you stored your files as well). Depending if you want to use the rgb values in the partition, adapt the call to libcp if need be.

If your files are huge and takes a really long time to be read, you can use the ```partition_semantic3d.py``` script instead, and change the calls to ```prune_labels``` in ```partiton/provider.py``` which process the files by blocks.

For the learning part you will need to make a script similar to ```learning/s3dis_dataset.py``` or ```learning/semantic3d_dataset.py``` depending if you have all the labels and want to validate by cross validation or if you have a classic training/validation set split. 

Modify ```learning/main.py``` L124 such that your script is called when using your dataset.

Finally adapt the argument of ```--model_config``` to your number of classes.

Let us know if you are successful or if more changes were needed!

We work with conda, but venv should work as well. Let us know if you encounter difficulties.

closed 2018-03-19 by loicland

loicland, 2018-03-20:
I have added a few methods which should simplify using other data sets.