loicland/superpoint_graph #9: STNkD final bias not trainable?

laoreja, 2018-03-06:
Hi, 
in https://github.com/loicland/superpoint_graph/blob/518fc08d7709a9f23a17a8e23c8f731f54bf81c8/learning/pointnet.py#L51 you are making the bias for the projection an Identity matrix and not trainable?

In the original pointnet implementation, they make this a bias term (https://github.com/charlesq34/pointnet/blob/d64d2398e55b24f69e95ecb549ff7d4581ffc21e/models/transform_nets.py#L49), which is trainable.

The STN is actually supposed to output a rotation matrix, so in my reasoning, the bias term should be trainable.

Do you make it untrainable on purpose, and why? 

Also, why the projection are initialized as all 0s. In deep learning courses, it's not encouraged to set the weights as the same value.

laoreja, 2018-03-06:
Also, why do you only do spatial transformer on xy dimensions, but not on xyz dimensions?

loicland, 2018-03-06:
Hi,

the biais is trainable, as a parameter of the proj layer ([doc on pytorch](http://pytorch.org/docs/master/nn.html#linear-layers)). We add the identity matrix afterwards, which amounts to the same thing as https://github.com/charlesq34/pointnet/blob/d64d2398e55b24f69e95ecb549ff7d4581ffc21e/models/transform_nets.py#L47

Projections are initialized at zero because we want to initialize the transform at identity.

We only transform on xy because we don't want the network to change the vertical orientation of superpoints. A wall and a ceiling have the same shape, barring their vertical orientation. Of course the network could learn not to rotate on z if need be, but we save parameters with this insight.

However, feel free to experiment with initilialization and feature transform and report potential improvements!

laoreja, 2018-03-06:
I see. Thank you very much for your quick reply!

closed 2018-03-06 by loicland