loicland/superpoint_graph #10: ImportError: No module named 'libcp'

post2web, 2018-03-07:
I get the import error when I try to run:
```
python partition/write_Semantic3d.py --SEMA3D_PATH $SEMA3D_DIR --odir "results/sema3d/best" --db_test_name testred
Traceback (most recent call last):
  File "partition/write_Semantic3d.py", line 20, in <module>
    import libcp
ImportError: No module named 'libcp'
```

post2web, 2018-03-07:
Fixed it by adding this:
```
sys.path.append("./partition/cut-pursuit/src")
sys.path.append("./partition/ply_c")
```
to write_Semantic3d.py

loicland, 2018-03-07:
Right, good catch, changing it now.

closed 2018-03-07 by loicland

WanpengShao, 2022-01-14:
Hi:

I have this in partition:
sys.path.append("/media/wpshao/D_Disk/ubuntu_code/partition/cut-pursuit/build/src")
sys.path.append("/media/wpshao/D_Disk/ubuntu_code/partition/ply_c")
sys.path.append("/media/wpshao/D_Disk/ubuntu_code/partition")
And I have compliled the libply_c and libcp and get libply_c.so and libcp.so.
But still:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'libcp'
How to fixed it?