loicland/superpoint_graph #12: About the grammar of pytorch

LZDSJTU, 2018-03-17:
I see that you use model.ecc and model.ptn during the model creating. I do not see such usage of nn.Module() before. I want to ask why can add .ecc and .ptn after variable 'model'. Is there any API or explanation for this grammar?

Thank you very much!

loicland, 2018-03-20:
ecc and .ptn are custom submodels of the modules that we create in ecc.py and pointnet.py

LZDSJTU, 2018-03-21:
yeap, I see the definition of the two modules in your code. If I change the model.ecc to model.abc or any other word, the code can also run properly. So I want to ask in pytorch : model = nn.module(), we can add any word after model?

LZDSJTU, 2018-03-21:
Also, about the structure of the network, I read your code and think that after pointnet, the embedding is transferred into a RNN. And the GRU cells are repeated for 10 times by the definition of gru_10. What I understand is that this is a conventional RNN which is connected one by one.
However, in your paper, according to figure 2(c), it seems that every superpoint embedding is connected with a GRU and some GRU cells are connected with each other. I do not see such structure of RNN. So can you provide any reference about this structure?
Thank you very much!

loicland, 2018-03-21:
The embeddings of superpoints are used to initialize the internal state of the GRU (```hx``` in ```/learning/modules/RNNGraphConvModule```. Then this internal state is updated 10 times with the input vector of the GRUs at each iteration  being computed from the adjacent GRUs.

The input vector of each superpoint at each iteration is defined as a weighted average of the incoming messages from adjacent superpoints (```/learning/ecc/GraphConvModule/GraphConvFunction```).  The incoming messages are defined as the product (matrix or hadamard, see method ```_multiply```) between the internal state of the GRU associated with the superpoint at the origin of the superedge and a weight matrix/vector characteristic of each superedge. This edge weight is computed by the edge filter generating network from the superedge features (see ```/learning/graphnet/create_fnet```).

If you want to read more about edge conditioned convolution networks, see [Martin's great paper](http://openaccess.thecvf.com/content_cvpr_2017/papers/Simonovsky_Dynamic_Edge-Conditioned_Filters_CVPR_2017_paper.pdf)

closed 2018-03-22 by loicland