loicland/superpoint_graph #13: Issues compiling ply_c library

siquike, 2018-03-17:
Hi,
Thank you so much for your wonderful work. I have struggled for weeks on trying to compile the ply_c and cut-pursuit libraries. It seems that anaconda is not allowing cmake to find the real paths both libraries need. How do you go around this? Have you experienced similar issues? Here is the output of cmake:

`cmake . -DPYTHON_LIBRARY=$CONDAENV/lib/libpython3.6m.so -DPYTHON_INCLUDE_DIR=$CONDAENV/include/python3.6m -DBOOST_INCLUDEDIR=$CONDAENV/include -DEIGEN3_INCLUDE_DIR=$CONDAENV/include/eigen3
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Warning at /usr/share/cmake-3.5/Modules/FindBoost.cmake:725 (message):
  Imported targets not available for Boost version 106501
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:763 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:1332 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:21 (find_package)

CMake Warning at /usr/share/cmake-3.5/Modules/FindBoost.cmake:725 (message):
  Imported targets not available for Boost version 106501
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:763 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:1332 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:21 (find_package)

-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   graph
--   numpy
-- Found PythonLibs: /lib/libpython3.6m.so  
PYTHON LIBRARIES ARE /lib/libpython3.6m.so
-- Configuring done
CMake Warning at src/CMakeLists.txt:3 (add_library):
  Cannot generate a safe linker search path for target cp because files in
  some directories may conflict with libraries in implicit directories:

    link library [libpython3.6m.so] in /lib may be hidden by files in:
      /home/siquike/anaconda3/lib

  Some of these libraries may not be found correctly.

CMake Warning at src/CMakeLists.txt:3 (add_library):
  Cannot generate a safe runtime search path for target cp because files in
  some directories may conflict with libraries in implicit directories:

    runtime library [libpython3.6m.so] in /lib may be hidden by files in:
      /home/siquike/anaconda3/lib

  Some of these libraries may not be found correctly.

-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    EIGEN3_INCLUDE_DIR
`

loicland, 2018-03-19:
Hi,

compilation can be tricky, especially if you have multiple versions of python and boost on your machine.
Some troubleshooting:
- it seems like cmake generating completes. What does the ```make``` say?
- replace directly in the call $CONDAENV with your conda environement :  ```/home/siquike/anaconda3/```
- change l15 of the CmakeList in ```ply_c``` to:
```find_package(Boost 1.65.0 COMPONENTS numpy3 python3 REQUIRED)``` if you are using python3
- remove ```boost_python``` l43

siquike, 2018-03-20:
This is what the make says:

```
/home/siquike/superpoint_graph/partition/ply_c/ply_c.cpp:4:23: fatal error: Eigen/Dense: No such file or directory
compilation terminated.
CMakeFiles/ply_c.dir/build.make:62: recipe for target 'CMakeFiles/ply_c.dir/ply_c.cpp.o' failed
make[2]: *** [CMakeFiles/ply_c.dir/ply_c.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ply_c.dir/all' failed
make[1]: *** [CMakeFiles/ply_c.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

I will try what you suggested.

siquike, 2018-03-20:
The replacement worked! Thank you very much!

siquike, 2018-03-20:
Last question. I am relatively new to some of the data alignment of these datasets. When I decompress the data using 7z, I get a txt file instead of the training and testing example individual files. How do I decompress the files in a way in which I get the individual files?

loicland, 2018-03-21:
Great! Would you be able to tell me which bullet fixed it?

Which data set are you referring to? for Semantic you have to download the .txt files individually, and the labels is just one zip. For Semantic you should have a single zip with 6 sbfolder if memory serves.

siquike, 2018-03-21:
Bullet 2 fixed it.

I am referring to Semantic. Yeah, so I download the individual zip files. The format is 7z. When I decompress each of them, each is a .txt file. Then, when I decompress the labels, I get multiple files with extension .label. Now, when I run the script, it says it cannot find files with extension .h5. How do I get the .h5 extension?
Also, for this part:

> For Semantic you should have a single zip with 6 sbfolder if memory serves.

With this, do you mean that after I unzip with 7z, do I need to recompress the files into a single zip file, with 6 different subdirectories within the zip file? Do I zip with 7z format?

loicland, 2018-03-21:
Hi,

you should have a ```data``` folder with subfloders ```train```, ```test_full``` and```test_reduced```. In each of this subfolders, you put the unzipped txt files. In the train you put the labels file as well. Your ROOT_PATH shall be the folder containing ```data```.

To obtain the corresponding h5 files (the features, superpointgraphs and parsed files) you need to partition, then preprocess.

And sorry, the 6 subfolder thing is for S3DIS.

zy20091082, 2018-03-22:
Sorry to bother you. I follow the instruction, but failed. @loicland 

In my situation, find_package can't work out to find     numpy3 python3

When I delete the line, it can complie, but comes out    "undefined symbol"

loicland, 2018-03-22:
Hi @zy20091082

Can you post the command line you use to for cmake?

What is the version of boost on your computer?

What is the output of ```locate libboost-python``` in a terminal?

tejaskhot, 2018-03-23:
Hi @loicland 

I am running into a similar issue with the very first `cmake` step where the wrong version of boost is detected. Following the bullet 2 above and replacing `$CONDAENV` with the absolute path does not seem to help resolve the issue. The output is : 
```
superpoint_graph/partition/ply_c$ cmake . -DPYTHON_LIBRARY=$/usr0/home/tkhot/anaconda3/lib/libpython3.6m.so -DPYTHON_INCLUDE_DIR=$/usr0/home/tkhot/anaconda3/include/python3.6m -DBOOST_INCLUDEDIR=$/usr0/home/tkhot/anaconda3/include -DEIGEN3_INCLUDE_DIR=$/usr0/home/tkhot/anaconda3/include/eigen3
CMake Error at /usr0/home/tkhot/anaconda3/share/cmake-3.9/Modules/FindBoost.cmake:1900 (message):
  Unable to find the requested Boost libraries.

  Boost version: 1.58.0

  Boost include path: /usr/include

  Detected version of Boost is too old.  Requested version was 1.65.1 (or
  newer).
Call Stack (most recent call first):
  CMakeLists.txt:15 (find_package)

Boost includes ARE /usr/include
Boost LIBRARIES ARE /usr/lib/x86_64-linux-gnu
PYTHON LIBRARIES ARE /usr0/home/tkhot/git_repos/superpoint_graph/partition/ply_c/$/usr0/home/tkhot/anaconda3/lib/libpython3.6m.so
-- Configuring incomplete, errors occurred!
See also "/usr0/home/tkhot/git_repos/superpoint_graph/partition/ply_c/CMakeFiles/CMakeOutput.log".
```
I have pasted the contents of `CMakeOutput.log` here: 
https://pastebin.com/zatdtRpv

tejaskhot, 2018-03-24:
@siquike How did you solve the issue with `make`? The second bullet i.e. using complete path for `$CONDAENV` does not seem to fix that issue for me

zy20091082, 2018-03-25:
Because of some other thing, sorry to reply late. @loicland 
It seems the system cannot find boost-python. But before I have install boost in Anaconda.

The output of locate libboost-python is:
```
/usr/share/doc/libboost-python-dev
/usr/share/doc/libboost-python1.58-dev
/usr/share/doc/libboost-python1.58.0
/usr/share/doc/libboost-python-dev/changelog.gz
/usr/share/doc/libboost-python-dev/copyright
/usr/share/doc/libboost-python1.58-dev/README.Debian
/usr/share/doc/libboost-python1.58-dev/changelog.Debian.gz
/usr/share/doc/libboost-python1.58-dev/copyright
/usr/share/doc/libboost-python1.58.0/changelog.Debian.gz
/usr/share/doc/libboost-python1.58.0/copyright
/usr/share/lintian/overrides/libboost-python1.58-dev
/usr/share/lintian/overrides/libboost-python1.58.0
/var/lib/dpkg/info/libboost-python-dev.list
/var/lib/dpkg/info/libboost-python-dev.md5sums
/var/lib/dpkg/info/libboost-python1.58-dev.list
/var/lib/dpkg/info/libboost-python1.58-dev.md5sums
/var/lib/dpkg/info/libboost-python1.58-dev.postinst
/var/lib/dpkg/info/libboost-python1.58-dev.prerm
/var/lib/dpkg/info/libboost-python1.58.0.list
/var/lib/dpkg/info/libboost-python1.58.0.md5sums
/var/lib/dpkg/info/libboost-python1.58.0.shlibs
/var/lib/dpkg/info/libboost-python1.58.0.triggers

```

loicland, 2018-03-25:
Hi,

It doesn't seem that anaconda has installed python, or you would have libboostpython in your coda environment.

Run ```conda install -c anaconda boost```

Also you need boost version 1.63 minimum.

amandaloh16, 2018-03-29:
Hi,
I am having the same issues: tried resolutions you suggest here and installed everything as you have it in the README: I keep getting this error: It cannot find the numpy library

amanda@amanda-Precision-Tower-5810:~/superpoint_graph/partition/ply_c$ cmake . -DPYTHON_LIBRARY=/home/amanda/anaconda3/lib/libpython3.6m.so -DPYTHON_INCLUDE_DIR=/home/amanda/anaconda3/include/python3.6m -DBOOST_INCLUDEDIR=/home/amanda/anaconda3/include -DEIGEN3_INCLUDE_DIR=/home/amanda/anaconda3/include/eigen3
CMake Warning at /usr/share/cmake-3.5/Modules/FindBoost.cmake:725 (message):
  Imported targets not available for Boost version 106501
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:763 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:1332 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:15 (find_package)

CMake Warning at /usr/share/cmake-3.5/Modules/FindBoost.cmake:725 (message):
  Imported targets not available for Boost version 106501
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:763 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:1332 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:15 (find_package)

-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   numpy3
--   python3
Boost includes ARE /home/amanda/anaconda3/include
Boost LIBRARIES ARE /home/amanda/anaconda3/lib
PYTHON LIBRARIES ARE /home/amanda/anaconda3/lib/libpython3.6m.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/amanda/superpoint_graph/partition/ply_c
amanda@amanda-Precision-Tower-5810:~/superpoint_graph/partition/ply_c$ make
[ 50%] Building CXX object CMakeFiles/ply_c.dir/ply_c.cpp.o
/home/amanda/superpoint_graph/partition/ply_c/ply_c.cpp:8:33: fatal error: numpy/ndarrayobject.h: No such file or directory
compilation terminated.
CMakeFiles/ply_c.dir/build.make:62: recipe for target 'CMakeFiles/ply_c.dir/ply_c.cpp.o' failed
make[2]: *** [CMakeFiles/ply_c.dir/ply_c.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ply_c.dir/all' failed
make[1]: *** [CMakeFiles/ply_c.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
amanda@amanda-Precision-Tower-5810:~/superpoint_graph/partition/ply_c$

loicland, 2018-03-29:
Hi amanda,

I think your version f CMake might be too old. Can you update it and try again?

If this doesn't work, can you post the results of these command lines in a terminal:
```locate anaconda3/lib/libpython``` 

And what is the version of boost in 
```conda list```

amandaloh16, 2018-03-29:
Here is the output of these commands for me

`amanda@amanda-Precision-Tower-5810:~$ cmake --version
cmake version 3.5.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).
amanda@amanda-Precision-Tower-5810:~$ locate anaconda3/lib/libpython
/home/amanda/anaconda3/lib/libpython3.5m.so
/home/amanda/anaconda3/lib/libpython3.5m.so.1.0
/home/amanda/anaconda3/lib/libpython3.so
amanda@amanda-Precision-Tower-5810:~$ conda list
# packages in environment at /home/amanda/anaconda3:
#
# Name                    Version                   Build  Channel
_ipyw_jlab_nb_ext_conf    0.1.0            py36he11e457_0  
alabaster                 0.7.10           py36h306e16b_0  
anaconda                  custom           py36hbbc8b67_0    anaconda
anaconda-client           1.6.9                    py36_0  
anaconda-navigator        1.7.0                    py36_0  
anaconda-project          0.8.2            py36h44fb852_0  
asn1crypto                0.24.0                   py36_0  
astroid                   1.6.1                    py36_0  
attrs                     17.4.0                   py36_0  
babel                     2.5.3                    py36_0  
backports                 1.0              py36hfa02d7e_1  
backports.shutil_get_terminal_size 1.0.0            py36hfea85ff_2  
beautifulsoup4            4.6.0            py36h49b8c8c_1  
bitarray                  0.8.1            py36h14c3975_1  
bleach                    2.1.2                    py36_0  
boost                     1.65.1                   py36_4    anaconda
boost                     0.1                       <pip>
boost-python              1.55.0                        1    meznom
boto                      2.48.0           py36h6e4cd66_1  
bzip2                     1.0.6                h9a117a8_4  
ca-certificates           2017.08.26           h1d4fec5_0  
cairo                     1.14.12              h77bcde2_0  
certifi                   2018.1.18                py36_0  
cffi                      1.11.4           py36h9745a5d_0  
chardet                   3.0.4            py36h0f667ec_1  
click                     6.7              py36h5253387_0  
cloudpickle               0.5.2                    py36_1  
clyent                    1.2.2            py36h7e57e65_1  
colorama                  0.3.9            py36h489cec4_0  
conda                     4.5.0                    py36_0  
conda-build               3.4.1                    py36_0  
conda-env                 2.6.0                h36134e3_1  
conda-verify              2.0.0            py36h98955d8_0  
contextlib2               0.5.5            py36h6c84a62_0  
cryptography              2.1.4            py36hd09be54_0  
cudatoolkit               8.0                           3  
cudnn                     7.0.5                 cuda8.0_0  
cupy                      2.5.0                     <pip>
curl                      7.58.0               h84994c4_0  
cycler                    0.10.0           py36h93f1223_0  
cython                    0.27.3           py36h1860423_0  
cytoolz                   0.9.0            py36h14c3975_0  
dask-core                 0.16.1                   py36_0  
dbus                      1.12.2               hc3f9b76_1  
decorator                 4.2.1                    py36_0  
distributed               1.20.2                   py36_0  
docutils                  0.14             py36hb0f60f5_0  
eigen                     3.3.3                h5721d7b_0  
eigen3                    3.2.1                         0    omnia
entrypoints               0.2.3            py36h1aec115_2  
et_xmlfile                1.0.1            py36hd6bccc3_0  
expat                     2.2.5                he0dffb1_0  
fastcache                 1.0.2            py36h14c3975_2  
fastrlock                 0.3                       <pip>
filelock                  2.0.13           py36h646ffb5_0  
flask                     0.12.2           py36hb24657c_0  
[... 172 lines cut ...]
tornado                   4.5.3                    py36_0  
tqdm                      4.19.9                    <pip>
traitlets                 4.3.2            py36h674d592_0  
transforms3d              0.3.1                     <pip>
typing                    3.6.2            py36h7da032a_0  
unicodecsv                0.14.1           py36ha668878_0  
unixodbc                  2.3.4                hc36303a_1  
urllib3                   1.22             py36hbe7ace6_0  
visdom                    0.1.7                     <pip>
wcwidth                   0.1.7            py36hdf4376a_0  
webencodings              0.5.1            py36h800622e_1  
werkzeug                  0.14.1                   py36_0  
wheel                     0.30.0           py36hfd4bba0_1  
widgetsnbextension        3.1.0                    py36_0  
wrapt                     1.10.11          py36h28b7045_0  
xlrd                      1.1.0            py36h1db9f0c_1  
xlsxwriter                1.0.2            py36h3de1aca_0  
xlwt                      1.3.0            py36h7b00a1f_0  
xz                        5.2.3                h55aa19d_2  
yaml                      0.1.7                had09818_2  
zeromq                    4.2.2                hbedb6e5_2  
zict                      0.1.3            py36h3a3bf81_0  
zlib                      1.2.11               ha838bed_2  
amanda@amanda-Precision-Tower-5810:~$ 
`

loicland, 2018-04-03:
Added the error source of this thread directly in the readme, closing the issue.

closed 2018-04-03 by loicland

tejaskhot, 2018-06-30:
@loicland I am getting the same error as @amandaloh16 above although my anaconda installation is using `python3.6m`. Here are the relevant outputs:

```
(pytorch3) tkhot@hartwell:/usr0/home/tkhot/superpoint_graph/partition/ply_c$ cmake . -DPYTHON_LIBRARY=/usr0/home/tkhot/anaconda3/envs/pytorch3/lib/libpython3.6m.so -DPYTHON_INCLUDE_DIR=/usr0/home/tkhot/anaconda3/envs/pytorch3/include/python3.6m -DBOOST_INCLUDEDIR=/usr0/home/tkhot/anaconda3/envs/pytorch3/include -DEIGEN3_INCLUDE_DIR=/usr0/home/tkhot/anaconda3/envs/pytorch3/include/eigen3
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   python
--   numpy
Boost includes ARE /usr0/home/tkhot/anaconda3/envs/pytorch3/include
Boost LIBRARIES ARE /usr0/home/tkhot/anaconda3/envs/pytorch3/lib
-- Found PythonLibs: /usr0/home/tkhot/anaconda3/envs/pytorch3/lib/libpython3.6m.so (found version "3.6.6") 
PYTHON LIBRARIES ARE /usr0/home/tkhot/anaconda3/envs/pytorch3/lib/libpython3.6m.so
-- Configuring done
-- Generating done
-- Build files have been written to: /usr0/home/tkhot/superpoint_graph/partition/ply_c
```

But then the `make` command fails.
```
(pytorch3) tkhot@hartwell:/usr0/home/tkhot/superpoint_graph/partition/ply_c$ make
Scanning dependencies of target ply_c
[ 50%] Building CXX object CMakeFiles/ply_c.dir/ply_c.cpp.o
/usr0/home/tkhot/superpoint_graph/partition/ply_c/ply_c.cpp:8:33: fatal error: numpy/ndarrayobject.h: No such file or directory
compilation terminated.
CMakeFiles/ply_c.dir/build.make:62: recipe for target 'CMakeFiles/ply_c.dir/ply_c.cpp.o' failed
make[2]: *** [CMakeFiles/ply_c.dir/ply_c.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ply_c.dir/all' failed
make[1]: *** [CMakeFiles/ply_c.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

Other details:
```
(pytorch3) tkhot@hartwell:/usr0/home/tkhot/superpoint_graph/partition/ply_c$ cmake --version
cmake version 3.9.4

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```
Checking `libpython`
```
(pytorch3) tkhot@hartwell:/usr0/home/tkhot/superpoint_graph/partition/ply_c$ locate anaconda3/lib/libpython 
/usr0/home/tkhot/anaconda3/lib/libpython3.6m.a
/usr0/home/tkhot/anaconda3/lib/libpython3.6m.so
/usr0/home/tkhot/anaconda3/lib/libpython3.6m.so.1
/usr0/home/tkhot/anaconda3/lib/libpython3.6m.so.1.0
```
However, it does not show up for the specific environment i.e. no output on running
```
locate anaconda3/envs/pytorch3/lib/libpython
```

tejaskhot, 2018-07-03:
I solved the issue by locating `numpy/ndarray.h` in the system. For me it was at `/usr0/home/tkhot/anaconda3/envs/pytorch3/lib/python3.6/site-packages/numpy/core/include/numpy`. I copied the entire directory to `/usr0/home/tkhot/anaconda3/envs/pytorch3/include/` which is where the make file was searching for it. 

On a side note, another common issue I faced was `fatal error: Eigen/Eigenvalues: No such file or directory`. This was fixed by editing `partition/ply_c/ply_c.cpp` and changing
```
#include <Eigen/Dense>
#include <Eigen/Eigenvalues>
```
to
```
#include <eigen3/Eigen/Dense>
#include <eigen3/Eigen/Eigenvalues>
```

loicland, 2018-07-03:
I did not have access to my computer to check your issue. Thanks for solving it yourself and posting the solution here!

loic