loicland/superpoint_graph #17: Question about the parameters in partition part

sycmio, 2018-04-05:
Hello Loic,

It's me again :D  Currently we are trying to apply your network on our dataset，and we found that the segmentation result is not good.  However, your partition part works pretty well and we plan to continue our work based on the partition result. So could you please tell me the exact meaning of the hyperparameters in partition part? I saw your help comments of these parameters, but some are a bit confusing to me.

Also, is it possible to only run the partition part and get the partition ply without running the network? Do we need a good GPU to run the partition part?

Looking forward to your reply!

Yongchi

loicland, 2018-04-06:
Hi,

if I remember correctly you are trying to directly apply the trained model from Semantic3D to your dataset am I correct? If so make sure that the point clouds is such that the road is about at zero elevation. In general I would strongly advise some fine-tuning with some labeled data, especially if your data set is not completely similar to the scenes in semantic3d.

You of course can only compute the partition. Just run the `partition/partition.py` function. To obtain the partition ply file just run 

```
python partition/visualize.py --dataset sema3d --ROOT_PATH $DIR  --file_path 'subfolder/file' --output_type p
```
or `--output_type ps` if you want to visualize the SPG, as explained in the updated README. The partition runs purely on the CPU, so you don't need a GPU at all.

As for the meta parameters, the main one is `--reg_strength`. play around a little but with it depending on the purity/granularity trade-off you are looking for.
`--lambda_edge_weight` determines the edge weight w.r.t the length. Increase the value if the scan lines are very spaced out.

sycmio, 2018-04-06:
Will try it. Thanks for your great work and kindly help!

closed 2018-04-09 by loicland