loicland/superpoint_graph #18: Partitioning based on geometric features only

Not-IITian, 2018-04-06:
Hi, 

Running your cut-pursuit purely on geometric features of point cloud, without any RGB, produces the following output:

Graph 1571 vertices and 37656 edges and observation of dimension 4
Iteration   1 -    1 components - Saturation 100.0 % - Quadratic Energy  nan % - Timer  0.025178
All components are saturated

I think this means the problem(graph) is ill conditioned so no iterations and thus, a degenerate solution? or maybe I am doing something silly.

Thanks for your reply.

loicland, 2018-04-06:
Hi,

first are you running the `partition.ply` method or something homemade?

It seems that  the quadratic energy is NaN. That could indicate that some of your obeservation are NaN, which would kill the process. Are you computing them yourself or using the methods in the git?

Not-IITian, 2018-04-06:
yes, running your source code partition.py with other functions in provider.py and graph.py, nothing homemade. I  check the graph input values to see if there is an NaN and get back.  Thanks

Not-IITian, 2018-04-09:
Value of lambda_0 in ply_c:compute_geof is 0 which makes the rest go NaN. Debugging python wrapper of C++ code is slow with print :)

loicland, 2018-04-09:
What point cloud are you applying the `compute_geof` function to? `lambda[0]=0` means that the spatial covariance matrix is 0, i.e. you have 45 points in the same exact position. Which is impossible since it should have been filtered out by the pruning.

maybe post here the point cloud (it seems very small? 1570 points) and I'll try to see what went wrong.

Not-IITian, 2018-04-09:
Thank you for your reply. I think it should help to resolve the issue else I will post the point cloud

loicland, 2018-04-20:
Updated the repo to make partitioning without rgb easier.

closed 2018-04-20 by loicland

ardianumam, 2023-04-26:
> What point cloud are you applying the `compute_geof` function to? `lambda[0]=0` means that the spatial covariance matrix is 0, i.e. you have 45 points in the same exact position. Which is impossible [...]

Hi,

I'm trying to partition [ShapeNet part dataset](https://www.researchgate.net/profile/Yongcheng-Liu/publication/332497055/figure/fig2/AS:748992291930112@1555585086201/Segmentation-examples-on-ShapeNet-part-benchmark.ppm) with number of points = 1024, and `libply_c.compute_geof` returns me NaN values. Any direction? Thanks.

loicland, 2023-04-26:
is it only nan values or just some?

ardianumam, 2023-04-26:
> is it only nan values or just some?

The returned `geof` has shape of [1024, 4], and the number of NaN elements is 4096 (which is 1024*4). So, all are NaN values.

loicland, 2023-04-26:
Can you try with this repo (part of the newer release of SPG planned in a couple weeks);

https://github.com/drprojects/point_geometric_features

ardianumam, 2023-04-26:
> Can you try with this repo (part of the newer release of SPG planned in a couple weeks);
> 
> https://github.com/drprojects/point_geometric_features

Thanks! Will try that out

loicland, 2023-04-26:
PS my student is updating the repo rihgt now to make it easier to compute the nn graph. Will let you know when done.

loicland, 2024-04-11:
Please post an issue on this repo `https://github.com/drprojects/point_geometric_features`  as their code solves a lots of the problem.

But typically, the NaN issues are due to the point cloud, eg points that are deduplicated, a NaN value somewhere, etc. I would start by inspecting it thoroughly.

wy9933, 2024-04-11:
> Please post an issue on this repo `https://github.com/drprojects/point_geometric_features` as their code solves a lots of the problem.
> 
> But typically, the NaN issues are due to the point cloud, eg points that are deduplicated, a NaN value somewhere, etc. I would start by inspecting it thoroughly.

Thanks, I will check out my data and code, sorry to bother. 😃