loicland/superpoint_graph #20: Can't file .hpy file

amandaloh16, 2018-04-09:
Was having an issue finding an /hpy file: realized a new trainval_best folder is generated even though the command: python partition/write_Semantic3d.py --SEMA3D_PATH $SEMA3D_DIR --odir "results/sema3d/best" --db_test_name testred specifies a different path in the folder 'best'

Found this command to work which implies predictions_testred.h5 is in trainval_best folder and not best folder.

amanda@amanda-Precision-Tower-5810:~/superpoint_graph$ python partition/write_Semantic3d.py --SEMA3D_PATH /home/amanda/Semantic3D/ --odir "results/sema3d/trainval_best" --db_test_name testred

Not sure if this is a problem other people are facing? If anyone  has this issue as well this is the solution I found.
'
amanda@amanda-Precision-Tower-5810:~/superpoint_graph$ python partition/write_Semantic3d.py --SEMA3D_PATH /home/amanda/Semantic3D/ --odir "results/sema3d/best" --db_test_name testred
/home/amanda/anaconda3/lib/python3.6/site-packages/h5py/__init__.py:36: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
=================
   test_reduced/
=================
Traceback (most recent call last):
  File "partition/write_Semantic3d.py", line 50, in <module>
    res_file = h5py.File(res_folder + 'predictions_' + args.db_test_name + '.h5', 'r')   
  File "/home/amanda/anaconda3/lib/python3.6/site-packages/h5py/_hl/files.py", line 269, in __init__
    fid = make_fid(name, mode, userblock_size, fapl, swmr=swmr)
  File "/home/amanda/anaconda3/lib/python3.6/site-packages/h5py/_hl/files.py", line 99, in make_fid
    fid = h5f.open(name, flags, fapl=fapl)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5f.pyx", line 78, in h5py.h5f.open
OSError: Unable to open file (unable to open file: name = './results/sema3d/best/predictions_testred.h5', errno = 2, error message = 'No such file or directory', flags = 0, o_flags = 0)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "partition/write_Semantic3d.py", line 52, in <module>
    raise ValueError("%s do not exists" % res_file) 
NameError: name 'res_file' is not defined
'

loicland, 2018-04-10:
Hi,
You should adapt ```--odir``` to chose the model you want, you can also rename them. I changed it in the instruction to make it so that ```write``` uses the inference from the model trained on the full train set (`train_val/best`) but feel free to rename them.

I am unsure I understand the rest of your message, do you have an error still? From what command?

amandaloh16, 2018-04-10:
No, this is more a note to anyone else because I spent a lot of time debugging the issue why I was getting OSErrors because I thought the write folder was default the same as the read folder later.

loicland, 2018-04-11:
I see. Thanks for pointing it out anyway, I changed the README so that it will be clearer.

closed 2018-04-11 by loicland