loicland/superpoint_graph #22: Label computation for SD3IS dataset

shubhtuls, 2018-04-24:
Hi,
Thanks a lot for the code release!

When reading the S3DIS data, the [code](https://github.com/loicland/superpoint_graph/blob/release/partition/provider.py#L162-L177) loads one 'object' at a time, and for each 'object vertex', assigns a label to its closest 'room vertex'. Should the nearest neighbor search not be done in the other direction i.e. for each 'room vertex', its label should be same as that of the closest 'object vertex' (across all objects)?

Thanks!

loicland, 2018-04-25:
Hi, thanks for carefully reading the code.

The trick here is that each object point is also a room point. You can check that the nearest neighbor is at a distance zero. In a way I use the octree structure of the room points as a hashtable to quickly find the matching object points. So doing what you suggest will yield the exact same result.

Advantage of doing my way:
- only compute one octree (the room one)
- some parts of the room are unlabelled (no corresponding objects), and we don't want to assign them any labels.

I hope this clears things up for you.

shubhtuls, 2018-04-25:
Hi,
Thanks for the reply and the clarification. I'd posted the issue as I found that some indices get written to multiple times, although after your comment I did check the distances and they are all 0. I think there might be some duplicate vertices which result in this.

As an example, when loading 'Area_5/storage_4', the object 'wall_4' had 145942 vertices and was being loaded 2nd in my execution. After all objects loaded, (room_object_indices == 1).sum() evaluates to 145213. Also, for this particular room, the number of room vertices was the same as the summation of the number of object vertices, implying that some room vertices were unassigned at the end.

I'm closing this issue since this  seems to only lead to missing a small number of vertices, and might not be a major concern.

closed 2018-04-25 by shubhtuls