loicland/superpoint_graph #23: The method mentioned in your paper

LZDSJTU, 2018-04-29:
In your paper: WEAKLY SUPERVISED SEGMENTATION-AIDED CLASSIFICATION OF URBAN SCENES FROM3D LIDAR POINT CLOUDS
You mentioned that you implemented two other methods: pointwise and CRF regularization.
Could you share the code? So I can compare the two methods.
Thank you!

loicland, 2018-04-30:
Hi,

the pointwise method was simply a random forest classifier trained from the same geometric features as the one computed in this repository by ```libply_c.compute_geof```.

The CRF methods are adapted from [this repository](https://github.com/loicland/point-cloud-regularization) (matlab wrappers only).

FYI these methods will massively underperform compared to superpoint graph or any modern approach of point cloud segmentation.

loicland, 2018-04-30:
closing this this is not related to this repo.

closed 2018-04-30 by loicland