loicland/superpoint_graph #24: Confusion matrix on custom dataset

maximiliangoettgens, 2018-04-30:
Hey Loic,

I am trying to train SPG on a 5-class labeled dataset (basically synthia: 360-degree-depthcams -> pointcloud)

I have the data in .ply format and as far as I can tell, loading of PC and labels as well as partitioning and SPG generation works. However, I am getting this error during training:
`Epoch 0/500 (results/synthia_model):
  0%|                                                                         | 0/1 [00:00<?, ?it/s]Traceback (most recent call last):
  File "learning/main.py", line 383, in <module>
    main()
  File "learning/main.py", line 286, in main
    acc, loss, oacc, avg_iou = train()
  File "learning/main.py", line 200, in train
    confusion_matrix.count_predicted_batch(tvec_cpu, np.argmax(o_cpu,1))
  File "/home/max/synthia_spg/learning/metrics.py", line 18, in count_predicted_batch
    self.confusion_matrix[:,predicted[i]] += ground_truth_vec[i,:]
IndexError: index 7 is out of bounds for axis 1 with size 5
`
I checked the confusion matrix and it is generated as 5x5 with zeros so that seems to work fine (it reads the 5 from custom_dataset.py).

1) The tvec_cpu matrix in the line mentioned above seems to be an one-hot representation of my labels, however instead of all 0 but one =1 the "hot" entry is some random integer. Is that correct?

2) The o_cpu matrix is a matrix of same row count as  tvec_cpu. However, it has a column count of 7 (0-6) and contains a bunch of values, mostly +-0.XXXXX - can you explain what this exactly is?

3) I think the problem lies somewhere within these passed arguments and the dimensionality of o_cpu compared to tvec_cpu.

I know this information is vague but any help is greatly appreciated since I am a little stuck here :D

Best, 
Max

maximiliangoettgens, 2018-04-30:
> It seems like your predicted value are not done with the correc number of classes. Which model configuration are you using? it should be gru_10,f_5 or gru_10_0,f_5 depending if you want matrix or vector edge filters (choose the later if your data set is small). The 5 indicates the number of classes.

Okay, I completely overlooked that one... It is training now and I will probably get back to you if I stumble across some other issues - I think there is something fishy with the z-scaling again but I will have to investigate that in greater detail, might also be some issue with the data itself - anyway thank you very much (again) for the great help!

loicland, 2018-04-30:
Right I completed the readme to make it more obvious.

For your z value make sure the ground is around 0. If you use a pre-trained model check the normalization in their dataset file and make sure to use the same.

closed 2018-04-30 by loicland

maximiliangoettgens, 2018-04-30:
Well, in a first try I used the original model trained on the Semantic3D dataset on my data which is in a metric scale just as semantic3D and KITTI. However, as a result I got only buildings and ground plane - as I increased the scaling to some 10000s value, the predictions for mid-height items were quite accurate. When testing on a different frame, I had to set the scaling parameter to some other fancy value to get meaningful results - both clouds hat their ground plane in z=0

But as I said don't take this to serious for now - it can easily be something in my data - I just checked whether there are any high values in the actual pcl files and there are none