loicland/superpoint_graph #25: Bug in provider.py:write_features

kushal-sharma, 2018-04-30:
On line 341:
    `data_file.create_dataset('rgb', data=rgb, dtype='uint8')`
On line 342:
```
    if len(rgb) > 0:
        data_file.create_dataset('rgb', data=rgb, dtype='uint8')
```
This causes problems while trying to save features_file.
```
Traceback (most recent call last):
  File "/home/sharma/miniconda3/envs/laser_int/lib/python3.6/pdb.py", line 1667, in main
    pdb._runscript(mainpyfile)
  File "/home/sharma/miniconda3/envs/laser_int/lib/python3.6/pdb.py", line 1548, in _runscript
    self.run(statement)
  File "/home/sharma/miniconda3/envs/laser_int/lib/python3.6/bdb.py", line 431, in run
    exec(cmd, globals, locals)
  File "<string>", line 1, in <module>
  File "/home/sharma/laser-integration/superpoint_graph/partition/partition.py", line 6, in <module>
    """
  File "partition/provider.py", line 343, in write_features
    data_file.create_dataset('rgb', data=rgb, dtype='uint8')
  File "/home/sharma/miniconda3/envs/laser_int/lib/python3.6/site-packages/h5py/_hl/group.py", line 109, in create_dataset
    self[name] = dset
  File "/home/sharma/miniconda3/envs/laser_int/lib/python3.6/site-packages/h5py/_hl/group.py", line 277, in __setitem__
    h5o.link(obj.id, self.id, name, lcpl=lcpl, lapl=self._lapl)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5o.pyx", line 202, in h5py.h5o.link
RuntimeError: Unable to create link (name already exists)
```

loicland, 2018-04-30:
Hi,

right, you have to manually erase the feature files if you want to write some new ones. I will changed it this week to make it automatic, thanks for pointing it out.

cheers,

loic

loicland, 2018-05-09:
Hi,

the latest commit should have fixed your issue. let me know if it is not the case,

loic