loicland/superpoint_graph #26: Training problem

dlliyang, 2018-05-02:
what's the matter? when i try CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset sema3d --SEMA3D_PATH $SEMA3D_DIR --db_test_name testred --db_train_name trainval \
--epochs 500 --lr_steps '[350, 400, 450]' --test_nth_epoch 100 --model_config 'gru_10,f_8' --ptn_nfeat_stn 11 \
--nworkers 2 --odir "results/sema3d/trainval_best".
-> Train accuracy: 62.34092131206309, 	Loss: 1.289891004562378
Epoch 99/500 (results/sema3d/trainval_best):
  0%|                                                                         | 0/5 [00:00<?, ?it/s]Ptn with 850 clouds
 20%|█████████████                                                    | 1/5 [00:00<00:03,  1.08it/s]Ptn with 858 clouds
 40%|██████████████████████████                                       | 2/5 [00:01<00:01,  1.51it/s]Ptn with 951 clouds
 60%|███████████████████████████████████████                          | 3/5 [00:01<00:01,  1.77it/s]Ptn with 677 clouds
 80%|████████████████████████████████████████████████████             | 4/5 [00:01<00:00,  2.05it/s]Ptn with 680 clouds
100%|█████████████████████████████████████████████████████████████████| 5/5 [00:02<00:00,  2.26it/s]
  0%|                                                                         | 0/4 [00:00<?, ?it/s]Ptn with 3655 clouds
/home/dilu/Downloads/superpoint_graph-release/learning/pointnet.py:133: UserWarning: volatile was removed and now has no effect. Use `with torch.no_grad():` instead.
  out = Variable(out.data, requires_grad=model.training, volatile=not model.training) # cut autograd
 25%|████████████████▎                                                | 1/4 [00:02<00:07,  2.34s/it]Ptn with 2887 clouds
 50%|████████████████████████████████▌                                | 2/4 [00:02<00:02,  1.33s/it]Ptn with 14142 clouds
THCudaCheck FAIL file=/pytorch/aten/src/THC/generic/THCStorage.cu line=58 error=2 : out of memory
Traceback (most recent call last):
  File "/home/dilu/Downloads/superpoint_graph-release/learning/main.py", line 387, in <module>
    main()
  File "/home/dilu/Downloads/superpoint_graph-release/learning/main.py", line 290, in main
    acc_test, oacc_test, avg_iou_test, avg_acc_test = eval()
  File "/home/dilu/Downloads/superpoint_graph-release/learning/main.py", line 225, in eval
    embeddings = ptnCloudEmbedder.run(model, *clouds_data)
  File "/home/dilu/Downloads/superpoint_graph-release/learning/pointnet.py", line 132, in run_full_monger
    out = model.ptn(Variable(clouds, volatile=True), Variable(clouds_global, volatile=True))
  File "/home/dilu/anaconda3/lib/python3.6/site-packages/torch/nn/modules/module.py", line 491, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/dilu/Downloads/superpoint_graph-release/learning/pointnet.py", line 94, in forward
    input = self.convs(input)
  File "/home/dilu/anaconda3/lib/python3.6/site-packages/torch/nn/modules/module.py", line 491, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/dilu/anaconda3/lib/python3.6/site-packages/torch/nn/modules/container.py", line 91, in forward
    input = module(input)
  File "/home/dilu/anaconda3/lib/python3.6/site-packages/torch/nn/modules/module.py", line 491, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/dilu/anaconda3/lib/python3.6/site-packages/torch/nn/modules/conv.py", line 176, in forward
    self.padding, self.dilation, self.groups)
RuntimeError: cuda runtime error (2) : out of memory at /pytorch/aten/src/THC/generic/THCStorage.cu:58

dlliyang, 2018-05-03:
thanks very much.my pytorch is 0.4.0,and when i change it to 0.2.0,it's ok now.thank you !

closed 2018-05-03 by dlliyang

loicland, 2018-05-03:
yes that's what I thought, 0.4.0 breaks the memory monger. I will fix it soon and let you know when it's done. Thanks for pointing it out.

it should work with 0.3.0 as well.