loicland/superpoint_graph #27: the accuracy and IoU for training dataset of Semantic3d

shuluoshu, 2018-05-05:
Hi, @loicland , thanks for your work, and when I want to refine the results of your model ,I generate the prediction results of training dataset of semantic3d by using your offered pretrained model, i.e., I use the modified **main.py** with cmd: `--db_test_name train`, I found that the prediction accuracy is much lower than your results on the benchmark.
 _Test accuracy: 85.96092934693056, 	Test oAcc: 0.6773388512039751, 	Test avgIoU: 0.45078531031139096, 	Test mAcc: 0.63300907_

I wonder why ? Thanks a lot.

loicland, 2018-05-05:
Hi,

Several things:
- the accuracy displayed in our paper is the one achieved on the reduced and full test, which can only be validated on the semantic3d.net servers
- as you can see [l25-35](https://github.com/loicland/superpoint_graph/blob/6f10d4105163c5899f212cc1338d76173c710fab/learning/sema3d_dataset.py#L25) of ```learning/sema3d_dataset.py```, `train` is not an option for `db_test_name`, only `test_red`, `test_full` and `val` a subset of 4 clouds from the training set used to validate our parameters.
- I am unsure what you want to do. Do you want the training error? In which case the easiest is probably to relaunch the learning for one extra epoch:

```CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset sema3d --SEMA3D_PATH $SEMA3D_DIR --db_test_name testred --db_train_name trainval \
--epochs 501 --lr_steps '[350, 400, 450]' --test_nth_epoch 100 --model_config 'gru_10,f_8' --ptn_nfeat_stn 11 --nworkers 2 --odir "results/sema3d/trainval_best_50```

shuluoshu, 2018-05-05:
@loicland , thanks for you immediate reply, 

- Yes, `train` is not an option for `db_test_name` in the original `main.py`, however, I modified this part in order to get the training results. 
- Actually, what I want is the predicted labels for **training** dataset because I will send them into another network for refinement, I just wonder why it performs worse than the reported results on test dataset. And, thanks, I will try your suggested method.

loicland, 2018-05-06:
Performing an extra epoch will not write the prediction for the training set, just compute the training error. I am unable to check before next Wednesday, let me know if you're still encountering weird behaviour by then and I'll have a look.

Loic

loicland, 2018-05-09:
Hi,

I had a few moments to check, and the training error of the trained model are as follow (after one extra iteration):
```Train accuracy: 83.82068164213787, 	Loss: 0.48701208404132296, 	Test accuracy: 91.60909702268142, 	Test oAcc: 0.9366899912898754, 	Test avgIoU: 0.794288625590343```