loicland/superpoint_graph #28: libcp library installation problem (`boost_numpy` not found)

mf27, 2018-05-08:
I get the following error when trying to compile the libcp library:
```
Could not find the following Boost libraries:

          boost_numpy

Some (but not all) of the required Boost libraries were found.  You may
  need to install these additional Boost libraries.  Alternatively, set
  BOOST_LIBRARYDIR to the directory containing Boost libraries or BOOST_ROOT
  to the location of Boost.
Call Stack (most recent call first):
  CMakeLists.txt:17 (find_package)
```
I installed the `Boost.Python` library (from [https://github.com/boostorg/python](https://github.com/boostorg/python)) but I get the same error.
Anyone with the same problem?

loicland, 2018-05-08:
Hi,

Which environment are you using? If anaconda does boost show up with ```conda list```?

Try ```locate libboostpython``` and change the link in the cmake options.

mf27, 2018-05-08:
Yes, anaconda. Boost is in the list. 
Ok, I simply removed the `CMakeCache.txt`, ran `cmake .` and `make`, and it compiled fine :). The problem was that I'm working in a conda environment so the paths were wrong.
Thanks for help, and for publishing the code.

closed 2018-05-08 by mf27