loicland/superpoint_graph #30: Semantic 3D for Training, Test reduced set custom

amandaloh16, 2018-05-09:
Hello,
I am trying to test some data I have, it is in x,y,z,i,i,i,i format to try and match the other reduced set .txts with x,y,z,i,r,g,b since I have no training data with labels so I must use Semantic3D for training. 

Can I just train with the Semantic 3D data but test with custom .txts in Semantic3D format. I am unclear still from the documentation how I can achieve this? Do I have to retrain all over again to just test my data?

loicland, 2018-05-10:
Hi,

You dont need to put empty values for i r g b if you do not have them. Simply write your loading function to output xyz (I already wrote the functions for .ply and .las format). Then follow the instructions in ```partition/partition.py``` for `custom_dataset`, I wrote examples of pruning without rgb nor labels.

You will need to retrain the models on semantic from scratch without rgb values (see [issue #6](https://github.com/loicland/superpoint_graph/issues/6) for the command lines).

Then adapt the beginning of ```learning/semantic3d_dataset``` to add a new test set.

Let me know if you encounter a problem!

Loic

amandaloh16, 2018-05-11:
So I wrote a custom format which is basically the same as semantic3D format minus anything rgb related, but I am confused because in training it's looking for a parsed folder but that's not produced in the partition step. Am I missing something?

amandaloh16, 2018-05-11:
So some follow up questions:
1. If I am to use same exact format as semantic3d but just add a new test file to the reduced set, how do I modify semantic3d_dataset.py? (My test file is .txt in same exact format of reduced set for Semantic3D). Do I have to just add the name of my new test files to the valid_names and everything else stays the same, or is there more to modify?

2. I made a brand new custom dataset method in provider.py, which is just a copy of the semantic3d set but just witholding rgb as you said, and filled out partition.py to call it and run the partition line: 

`python partition/partition.py --dataset custom_dataset --ROOT_PATH /home/amanda/Semantic3D --voxel_width 0.05 --reg_strength 0.8 --ver_batch 5000000`

However, this command is taking way longer than expected and when I have run it in the past: like >2hrs: is this normal?

amandaloh16, 2018-05-30:
What I'm trying to clarify which is the source of the dimension issue is: in sem3d_dataset.py:

the line:  P = np.concatenate([xyz, elpsv], axis=1)
am I supposed to omit the rgb value? My rgb is now an empty list when we make the changes you said to specify above: I think the line is originally:
line 95:  P = np.concatenate([xyz, rgb, elpsv], axis=1)

This one line code change is what is causing the dimension issues throughout the rest of the code.

loicland, 2018-05-31:
You are right, it is the incriminating line. Yopu can eiher:
- fill the rgb values with zeros (inefficient but easy)
- change line 212 in `learning/spg.py` to reflect that the columns are different:
```
if 'xyz' in args.pc_attribs: columns.append(P[:,:3])
if 'e' in args.pc_attribs: columns.append(P[:,3,None])
if 'lpsv' in args.pc_attribs: columns.append(P[:,4:7])
```

amandaloh16, 2018-05-31:
Yes! I found the line, thanks resolved.

closed 2018-05-31 by loicland