loicland/superpoint_graph #31: Pytorch 0.4 (I believe) prevents results from being written

lostcolony, 2018-05-11:
On line 299 in learning/main.py, a call is made to json.dump(), which is causing an error of "TypeError: Object of type 'Tensor' is not JSON serializable". A little searching and I've seen one or two other errors indicating that the upgrade to Pytorch 0.4 is responsible.

Calling str() on the tensor first will cause it to write out in a readable format. If that's the only goal (I don't know what the original call to json.dump did, other than, obviously, represented it as JSON. I don't know if it's the same representation), I can make a pull request; if there are other expectations on that output (it may show up when I get to those steps) I can take a stab at them, but it may warrant additional attention.

Also, just a warning, but future proofing (so either lock down the Pytorch version or fix; easy fix), getting a few deprecation warnings related to accessing loss.data[0] -

learning/main.py:196: UserWarning: invalid index of a 0-dim tensor. This will be an error in PyTorch 0.5. Use tensor.item() to convert a 0-dim tensor to a Python number
  loss_meter.add(loss.data[0])
learning/main.py:202: UserWarning: invalid index of a 0-dim tensor. This will be an error in PyTorch 0.5. Use tensor.item() to convert a 0-dim tensor to a Python number
  logging.debug('Batch loss %f, Loader time %f ms, Trainer time %f ms.', loss.data[0], t_loader, t_trainer)

loicland, 2018-05-15:
Good catch. Indeed our code is not 0.4 proof yet. For now I just changed the README warning not to use 0.4.

If you have the time to make a pull request I will look into it, else I will just do it by the end of may.

loicland, 2018-05-24:
We tested it on 0.2 0.3.0 and 0.3.1, so you should go with 0.3.1.