loicland/superpoint_graph #32: Inference on custom indoor dataset with S3DIS-trained model  : fnet weights don't agree

DamGuiVin, 2018-05-15:
Hello, 

First a big thank you for sharing this impressive work and working hard on making it reusable!

I have been trying to recover your trained S3DIS best models and test them for inference on a dataset of mine. Hence, after adapting the recommended files in /partition and /learning, I first successfully run the unsupervized partition on my _custom_indoor_data_ with : 

`python partition/partition.py --dataset custom_indoor --ROOT_PATH $INDOOR_DATA --voxel_width 0.03 --reg_strength 0.03`

Then I runI the SPG segmentation with :   

`for FOLD in 1 2 3 4 5 6; do CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset custom_indoor --CUSTOM_INDOOR_PATH $INDOOR_DATA --cvfold $FOLD --epochs -1 --lr_steps '[275,320]' --test_nth_epoch 50 --model_config 'gru_10_0,f_13' --ptn_nfeat_stn 14 --nworkers 2 --odir "models/s3dis/cv${FOLD}" --resume RESUME; done`

However, when loading the weights from the checkpoint, there seems to be a disagreement in the output dimension of the the 7th linear layer of the fnet :  

`RuntimeError: While copying the parameter named ecc.0._fnet.7.weight, whose dimensions in the model are torch.Size([32, 64]) and whose dimensions in the checkpoint are torch.Size([1024, 64])`

Would you have an idea of what may be causing this ? Have I poorly parameterized my inference ? Are the checkpoint weights provided not the final ones ?

Thanks in advance for the tips !

loicland, 2018-05-15:
Ah, that's my bad, I changed the naming convention for the --model_config arguments and did not think how that would impact the loading of trained models. I (quick)fixed it, it should work now, get the last commit it should work now. Tell me if otherwise!

loic

DamGuiVin, 2018-05-16:
I used your patch and could succesfully load the pretrained model weights.
Thanks for the help and quick reply !

closed 2018-05-16 by DamGuiVin