loicland/superpoint_graph #33: Questions about labels_sub

penghao1990, 2018-05-17:
Thanks for your work!
Why labels_sub is a N X (Class+1) dimensional matrix. It's a simple question, but I don't really understand it.
thank you!

loicland, 2018-05-17:
When pruning, 3D points are clustered in voxels within a regular grid. In turn, each of these voxels are  represented by a 3D point whose position and color are the average of the points it contains. However, its label is represented by a vector containing the break down of labels of the points it contains (i.e. the number of points of each label).

Furthermore, we add the class '0' for unlabelled points.

Let me know if I am not being clear.

loic

penghao1990, 2018-05-18:
thanks for your help

closed 2018-05-18 by loicland