loicland/superpoint_graph #34: Superedge features with Naan values

DamGuiVin, 2018-05-17:
Hello !

Disclaimer : this is not really a question, rather a warning for an issue I came accross when applying the s3dis-pretrained model on my own data for inference and that someone else might encounter.

It so happened that `preprocessing.StandardScaler().fit(edge_feats)` in `learning/spg.py` would throw a "Nan or infinite or non-float32" error when trying to scale the superedge features.

After investigating a little, I found that this was caused by the fact that the 3rd eigenvalue of the PCA of superpoints in `compute_sp_graph()` of `partition/graphs.py` may be slightly negative, due to numerical instability for skewed superpoint shapes I presume. This in return may affect the computation of the `sp_volume` feature at `graph["sp_volume"][i_com] = np.sqrt(ev[0] * ev[1] * ev[2] + 1e-10)`. If the product of the eigenvalues is negative enough, the sqrt returns a Nan without throwing an error.

To overcome this, I just changed `ev = -np.sort(-ev[0])` into `ev = -np.sort(-np.abs(ev[0]))`, to ensure all PCA eigenvalues are positive.

DamGuiVin, 2018-05-18:
Yes I thought so too but it seems that when computing the superpoint volume feature, if there is 1 negative eigenvalue and 2 other larger positive ones, the product of the eigenvalues 'take over' the `+1e-10`

DamGuiVin, 2018-05-18:
I close this issue so it does not clog your "issues" area

closed 2018-05-18 by DamGuiVin

loicland, 2018-05-18:
ah yes good catch, did not think of this figure case. Will add your fix as soon as I have time to check it does not impact the quality of the results on trained models.