loicland/superpoint_graph #36: Issue with building libply_c

DrFridolin, 2018-06-04:
Hello,

I am following the step-by-step build instructions and I am running into a problem when trying to build libply_c. More specifically, when I run 'make', I get the following error message:

[ 50%] Building CXX object CMakeFiles/ply_c.dir/ply_c.cpp.o
~/partition/ply_c/ply_c.cpp:8:33: fatal error: numpy/ndarrayobject.h: No such file or directory
compilation terminated.
CMakeFiles/ply_c.dir/build.make:62: recipe for target 'CMakeFiles/ply_c.dir/ply_c.cpp.o' failed
make[2]: *** [CMakeFiles/ply_c.dir/ply_c.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ply_c.dir/all' failed
make[1]: *** [CMakeFiles/ply_c.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

As far as I can tell, numpy is installed and ndarrayobject.h is present in two locations:

~/anaconda3/lib/python3.6/site-packages/numpy/core/include/numpy/ndarrayobject.h
~/anaconda3/pkgs/numpy-1.14.0-py36h3dfced4_1/lib/python3.6/site-packages/numpy/core/include/numpy/ndarrayobject.h

Do I need to edit the CMakeLists.txt file somehow? Do you have an idea what might be causing this issue?

Thanks

DrFridolin, 2018-06-05:
When I do locate anaconda3/lib/libpython, I get the following results:
~/anaconda3/lib/libpython3.6m.a
~/anaconda3/lib/libpython3.6m.so
~/anaconda3/lib/libpython3.6m.so.1
~/anaconda3/lib/libpython3.6m.so.1.0

I get the following error when I change 3.6m to 3.6 in the two arguments in CMake command line:

[ 50%] Building CXX object CMakeFiles/ply_c.dir/ply_c.cpp.o
In file included from ~/anaconda3/include/boost/python/detail/prefix.hpp:13:0,
                 from ~/anaconda3/include/boost/python/args.hpp:8,
                 from ~/anaconda3/include/boost/python.hpp:11,
                 from ~/repositories/SuperPointGraphs/partition/ply_c/ply_c.cpp:6:
~/anaconda3/include/boost/python/detail/wrap_python.hpp:50:23: fatal error: pyconfig.h: No such file or directory
compilation terminated.
CMakeFiles/ply_c.dir/build.make:62: recipe for target 'CMakeFiles/ply_c.dir/ply_c.cpp.o' failed
make[2]: *** [CMakeFiles/ply_c.dir/ply_c.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ply_c.dir/all' failed
make[1]: *** [CMakeFiles/ply_c.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

ArthurOuaknine, 2018-06-11:
I had the same initial issue than you @DrFridolin (ndarrayobject.h not found).
I solved it by copying the numpy library in the right place (for me).
I think you just have to use the following command line:
'''
cp -r ~/anaconda3/lib/python3.6/site-packages/numpy/core/include/numpy /usr/local/include
'''
Then you have to clone again the repository and use the cmake command line of the tutorial.

I also had the same second error you have (pyconfig.h not found). It is just because my  '''$PATH''' was not at the correct place. I think your settings were correct when you had the first error and you should not change them. 
If you still have this second problem, run this command line:
'''
locate pyconfig.h
'''
And you will have to correct path to use.

Unfortunately, I am now stuck in another error still while building libply_c:
'''
/Users/arthurouaknine/Documents/work/faircast/3d_points/superpoint_graph/partition/ply_c/ply_c.cpp:55:29: error: 
      arithmetic on a pointer to void
            memcpy(arr_data + i * dims[1] * cell_size, &(vecvec[i][0]), dims[1] * cell_size);
                   ~~~~~~~~ ^
1 error generated.
make[2]: *** [CMakeFiles/ply_c.dir/ply_c.cpp.o] Error 1
make[1]: *** [CMakeFiles/ply_c.dir/all] Error 2
make: *** [all] Error 2
'''
I have only changed the CMakeLists.txt file to set compilers:
'''
set(CMAKE_C_COMPILER "/usr/local/Cellar/llvm/6.0.0/bin/clang")
set(CMAKE_CXX_COMPILER "/usr/local/Cellar/llvm/6.0.0/bin/clang++")
'''
Is someone had the same error ? It is maybe related to the clang compiler but it didn't worked when I used gcc-8.

Thanks.

nicolas-chaulet, 2018-10-19:
Hi! I just ran into the same issue. I fixed it by adding the numpy include directory to the `CMakeLists.txt`. Would that be something to add to this file by default?

lapetite123, 2019-04-19:
I got the same error as DrFridolin when trying to build libply_c
 I tried to add  'set(PYTHON_INCLUDE_DIRS ${PYTHON_INCLUDE_DIRS} ~/anaconda3/lib/python3.6/site-packages/numpy/core/include/' in CMakeLists.txt file right before INCLUDE_DIRECTORIES(${PYTHON_INCLUDE_DIRS}) ,it didn't work for me.

I also tried ' cp -r ~/anaconda3/lib/python3.6/site-packages/numpy/core/include/numpy /usr/local/include',however permission denied. 
Is there another way to solve the problem?

lapetite123, 2019-04-20:
@nicolas-chaulet  can you please tell me what you exactly add in CMakeLists.txt ?

loicland, 2019-04-20:
Hi,

Have you checked the issue #59 ?

I assume @nicolas-chaulet made a direct reference to ${NUMPY_INCLUDE_DIRS} in the `INCLUDE_DIRECTORIES`

lapetite123, 2019-04-20:
Hi，I have checked the issue59, it might not help
I have changed the CMakeList.txt by adding ' set(PYTHON_INCLUDE_DIRS ${PYTHON_INCLUDE_DIRS} ~/home/vi/miniconda3/envs/abc/lib/python3.6/site-packages/numpy/core/include/)'  before 'INCLUDE_DIRECTORIES(${PYTHON_INCLUDE_DIRS})' , it didn't work.

[ 50%] Building CXX object CMakeFiles/ply_c.dir/ply_c.cpp.o
/home/vi/superpoint_graph/partition/ply_c/ply_c.cpp:8:33: fatal error: numpy/ndarrayobject.h: No such file or directory
compilation terminated.
CMakeFiles/ply_c.dir/build.make:62: recipe for target 'CMakeFiles/ply_c.dir/ply_c.cpp.o' failed
make[2]: *** [CMakeFiles/ply_c.dir/ply_c.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ply_c.dir/all' failed
make[1]: *** [CMakeFiles/ply_c.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

loicland, 2019-04-20:
I meant 

```
INCLUDE_DIRECTORIES(
${PYTHON_INCLUDE_DIRS}
${NUMPY_INCLUDE_DIRS} 
}
```

lapetite123, 2019-04-23:
After adding INCLUDE_DIRECTORIES(
${PYTHON_INCLUDE_DIRS}
${NUMPY_INCLUDE_DIRS} 
) into CMakeList, but it occurs another problem

lapetite123, 2019-04-23:
After installing annaconda3 instead of miniconda3 and adding numpy include direction to CMakeLists.txt, it can be built with several warning. Thanks very much for your help!