loicland/superpoint_graph #37: Getting all-zero labels for reduced test set with downloaded pre-trained model

DrFridolin, 2018-06-05:
Hello,

I am trying to get some segmentation results on the Semantic3D reduced test set.

I have downloaded the pre-trained model from the following link, renamed it model.pth.tar and placed it under /results/sema3d/trainval_best
http://imagine.enpc.fr/~simonovm/largescale/model_sema3d_trainval.pth.tar

I have run partition.py on the reduced test set.

I then run the following two commands:

python learning/sema3d_dataset.py --SEMA3D_PATH $SEMA3D_DIR

CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset sema3d --SEMA3D_PATH $SEMA3D_DIR --db_test_name testred --db_train_name trainval --epochs 500 --lr_steps '[350, 400, 450]' --test_nth_epoch 100 --model_config 'gru_10,f_8' --ptn_nfeat_stn 11 --nworkers 2 --odir "results/sema3d/trainval_best" --resume RESUME

After the second one, I get the following warning:

100%|█████████████████████████████████████████████████████████████████| 4/4 [00:14<00:00,  3.69s/it]
100%|█████████████████████████████████████████████████████████████████| 4/4 [00:14<00:00,  3.64s/it]
100%|█████████████████████████████████████████████████████████████████| 4/4 [00:15<00:00,  3.83s/it]
100%|█████████████████████████████████████████████████████████████████| 4/4 [00:14<00:00,  3.71s/it]
100%|█████████████████████████████████████████████████████████████████| 4/4 [00:14<00:00,  3.69s/it]
100%|█████████████████████████████████████████████████████████████████| 4/4 [00:14<00:00,  3.74s/it]
100%|█████████████████████████████████████████████████████████████████| 4/4 [00:15<00:00,  3.79s/it]
100%|█████████████████████████████████████████████████████████████████| 4/4 [00:14<00:00,  3.73s/it]
100%|█████████████████████████████████████████████████████████████████| 4/4 [00:14<00:00,  3.75s/it]
100%|█████████████████████████████████████████████████████████████████| 4/4 [00:14<00:00,  3.73s/it]
~/repositories/SuperPointGraphs/learning/metrics.py:66: RuntimeWarning: invalid value encountered in true_divide
  return sum(values) / class_seen
-> Multisample 10: Test accuracy: 100.0, 	Test oAcc: 0.0, 	Test avgIoU: nan, 	Test mAcc: 0.0

After this point, when I write the labels, I see that they are all zero, and the resulting prediction point cloud has a single grey color.

Can you see what I'm doing wrong in this process?

Thanks for your help.

DrFridolin, 2018-06-05:
When I try to train the model myself, by omitting the resume argument, I get the following:

Epoch 0/500 (results/sema3d/trainval_best):
100%|█████████████████████████████████████████████████████████████████| 7/7 [00:06<00:00,  1.09it/s]
/home/local/SRI/e28454/repositories/SuperPointGraphs/learning/metrics.py:66: RuntimeWarning: invalid value encountered in true_divide
  return sum(values) / class_seen
-> Train accuracy: 80.78873732074152, 	Loss: 0.655669653023194
Epoch 1/500 (results/sema3d/trainval_best):
100%|█████████████████████████████████████████████████████████████████| 7/7 [00:05<00:00,  1.20it/s]
-> Train accuracy: 100.0, 	Loss: 5.94529438607034e-06
Epoch 2/500 (results/sema3d/trainval_best):
100%|█████████████████████████████████████████████████████████████████| 7/7 [00:05<00:00,  1.27it/s]
-> Train accuracy: 100.0, 	Loss: 6.107642467968941e-08
Epoch 3/500 (results/sema3d/trainval_best):
100%|█████████████████████████████████████████████████████████████████| 7/7 [00:05<00:00,  1.34it/s]
-> Train accuracy: 100.0, 	Loss: 0.0
Epoch 4/500 (results/sema3d/trainval_best):
100%|█████████████████████████████████████████████████████████████████| 7/7 [00:05<00:00,  1.26it/s]
-> Train accuracy: 100.0, 	Loss: 0.0

The rest of the epochs are identical in that train accuracy shows 100.0 and loss shows 0.0 for each one.

loicland, 2018-06-05:
Hi,

The test sets of semantic 3d are unannotated, as you need to submit on their website to get a test score.

Consequently the label field is all zeros and you can't learn with it. You should use the training set to learn your models, or our split into training set / validation set to validate your changes.

To write the prediction clouds you should use the `visualize` function with `--outputype p'. If the `_pred.ply` in `/clouds` is grey, then we have a problem.

Does the reduced_test folder contains a prediction_testred.h5 file? If so, can you check it's values?

DrFridolin, 2018-06-05:
Hi Loic,

First off, thank you very much for your prompt responses.

I know that test sets are not annotated. I was just trying to produce some segmentation results on the reduced test set to look at it visually. I tried using the pre-trained model with the --resume setting, and I also tried training my own model from scratch. I assume I should be able to do these.

When I run learning/main.py and then partition/write_Semantic3d.py with the suggested parameters, I get .label files under $SEMA3D_DIR/labels that have all 0s in them and nothing else. I assume that's why the generated visualization is all grey.

When calling artition/visualize.py, I use --output_type ifprs as suggested.

There is a predictions_testred.h5 file under results, and it has the following values:

8948 4446 0d0a 1a0a 0000 0000 0008 0800
0400 1000 0000 0000 0000 0000 0000 0000
ffff ffff ffff ffff d8b4 0400 0000 0000
ffff ffff ffff ffff 0000 0000 0000 0000
6000 0000 0000 0000 0100 0000 0000 0000
8800 0000 0000 0000 a802 0000 0000 0000
0100 0100 0100 0000 1800 0000 0000 0000
1100 1000 0000 0000 8800 0000 0000 0000
a802 0000 0000 0000 5452 4545 0000 0100
ffff ffff ffff ffff ffff ffff ffff ffff
0000 0000 0000 0000 3065 0000 0000 0000
0800 0000 0000 0000 0000 0000 0000 0000

It has all zeros after these lines.

Another thing that might be related:
I was initially getting an "incorrectly setup Cupy" error when I was running /learning/main.py. When I installed "cupy-cuda90" instead of just "cupy" with pip install, the error went away. Could this be somehow related? I have cuda 9.0 installed.

loicland, 2018-06-05:
I meant to look at the value with a h5 reader,  like h5py or hview:)

I use cuda9, but would be very surprised if cuda9 was the culprit here.

Does the partition step go well? Is the partition file sensical? Can you run the inference in debug mode and check the values of embeddings and output in `eval_final`?

DrFridolin, 2018-06-05:
The partition step seems to be going well, and I can visualize the color-coded ply file with the computed partitions. Attached is a screenshot. I think the partitions look sensible.

[image]

When you say "run inference in debug mode", you mean running /learning/main.py in debug mode with some sort of IDE? Should I check the values of the outputs in the line below, or something else?

acc_test, oacc_test, avg_iou_test, per_class_iou_test, predictions_test, avg_acc_test, confusion_matrix = eval_final()

Regarding cuda version: Can we even use pytorch 0.3.1 with cuda 9.0? I was following the instructions here: https://pytorch.org/previous-versions/ but 

conda install pytorch=0.3.1 cuda90 -c soumith

gives me the following error:

Solving environment: failed
PackagesNotFoundError: The following packages are not available from current channels:
  - cuda90

So I actually ended up installing pytorch with:

conda install pytorch=0.3.1 cuda80 -c soumith

even though I have cuda 9.0 installed. Could this be related?

loicland, 2018-06-06:
I can't test the different versions right now, but I would be surprised if that's the problem.

Ideally yes, you would run the code in an IDE in debug mode, put a stop point line 256 or `learning/main.py` and run the inference.

By looking at  clouds_data (containing the superpoints), embeddings (containing the pointnet embeddings) and outputs (containing the output of the graph convolutional network) we would be able to see what went wrong hopefully!

DrFridolin, 2018-06-07:
I'm available to get you the things you need, so no need to wait for monday as long as you are available to reply :)

When I try to train the model from scratch with that command, I get the following warning after the 1st epoch:

/home/local/SRI/e28454/repositories/SuperPointGraphs/learning/metrics.py:66: RuntimeWarning: invalid value encountered in true_divide
  return sum(values) / class_seen
-> Train accuracy: 80.17628911414721, 	Loss: 0.6746747928887739

After that, the 3 epochs output the following, and train accuracy and loss remains the same for the rest of the epochs and training finishes relatively quickly. The segmentation results are all-grey in this case as well.

-> Train accuracy: 80.17628911414721, 	Loss: 0.6746747928887739
Epoch 1/500 (results/sema3d/trainval_best):
100%|█████████████████████████████████████████████████████████████████| 7/7 [00:05<00:00,  1.34it/s]
-> Train accuracy: 100.0, 	Loss: 1.1066283734960183e-05
Epoch 2/500 (results/sema3d/trainval_best):
100%|█████████████████████████████████████████████████████████████████| 7/7 [00:05<00:00,  1.32it/s]
-> Train accuracy: 100.0, 	Loss: 1.2826265205490017e-07
Epoch 3/500 (results/sema3d/trainval_best):
100%|█████████████████████████████████████████████████████████████████| 7/7 [00:05<00:00,  1.36it/s]
-> Train accuracy: 100.0, 	Loss: 0.0

Below is the output of print(model). I think this output is the same whether I try to load the existing model, or train one from scratch.

Module(
  (ecc): GraphNetwork(
    (0): RNNGraphConvModule(
      (_cell): GRUCellEx(32, 32)(ingate layernorm)
      (_fnet): Sequential(
        (0): Linear(in_features=13, out_features=32, bias=True)
        (1): ReLU(inplace)
        (2): Linear(in_features=32, out_features=128, bias=True)
        (3): ReLU(inplace)
        (4): Linear(in_features=128, out_features=64, bias=True)
        (5): BatchNorm1d(64, eps=1e-05, momentum=0.1, affine=True)
        (6): ReLU(inplace)
        (7): Linear(in_features=64, out_features=32, bias=False)
      )
    )
    (1): Linear(in_features=352, out_features=8, bias=True)
  )
  (ptn): PointNet(
    (stn): STNkD(
      (convs): Sequential(
        (0): Conv1d(11, 64, kernel_size=(1,), stride=(1,))
        (1): BatchNorm1d(64, eps=1e-05, momentum=0.1, affine=True)
        (2): ReLU(inplace)
        (3): Conv1d(64, 64, kernel_size=(1,), stride=(1,))
        (4): BatchNorm1d(64, eps=1e-05, momentum=0.1, affine=True)
        (5): ReLU(inplace)
        (6): Conv1d(64, 128, kernel_size=(1,), stride=(1,))
        (7): BatchNorm1d(128, eps=1e-05, momentum=0.1, affine=True)
        (8): ReLU(inplace)
      )
      (fcs): Sequential(
        (0): Linear(in_features=128, out_features=128, bias=True)
        (1): BatchNorm1d(128, eps=1e-05, momentum=0.1, affine=True)
        (2): ReLU(inplace)
        (3): Linear(in_features=128, out_features=64, bias=True)
        (4): BatchNorm1d(64, eps=1e-05, momentum=0.1, affine=True)
        (5): ReLU(inplace)
      )
      (proj): Linear(in_features=64, out_features=4, bias=True)
    )
    (convs): Sequential(
      (0): Conv1d(11, 64, kernel_size=(1,), stride=(1,))
      (1): BatchNorm1d(64, eps=1e-05, momentum=0.1, affine=True)
      (2): ReLU(inplace)
      (3): Conv1d(64, 64, kernel_size=(1,), stride=(1,))
      (4): BatchNorm1d(64, eps=1e-05, momentum=0.1, affine=True)
      (5): ReLU(inplace)
      (6): Conv1d(64, 128, kernel_size=(1,), stride=(1,))
      (7): BatchNorm1d(128, eps=1e-05, momentum=0.1, affine=True)
      (8): ReLU(inplace)
      (9): Conv1d(128, 128, kernel_size=(1,), stride=(1,))
      (10): BatchNorm1d(128, eps=1e-05, momentum=0.1, affine=True)
      (11): ReLU(inplace)
      (12): Conv1d(128, 256, kernel_size=(1,), stride=(1,))
      (13): BatchNorm1d(256, eps=1e-05, momentum=0.1, affine=True)
      (14): ReLU(inplace)
    )
    (fcs): Sequential(
      (0): Linear(in_features=257, out_features=256, bias=True)
      (1): BatchNorm1d(256, eps=1e-05, momentum=0.1, affine=True)
      (2): ReLU(inplace)
      (3): Linear(in_features=256, out_features=64, bias=True)
      (4): BatchNorm1d(64, eps=1e-05, momentum=0.1, affine=True)
      (5): ReLU(inplace)
      (6): Linear(in_features=64, out_features=32, bias=True)
    )
  )
)

DrFridolin, 2018-06-07:
Hi,

I put a breakpoint on line 180 in main.py and the contents of label_mode and label_vec seem to be all zeros, and they do not change throughout the training process:

label_mode:
 0
 0
 0
⋮ 
 0
 0
 0
[torch.cuda.LongTensor of size 1568 (GPU 0)]

label_vec:
    0     0     0  ...      0     0     0
    0     0     0  ...      0     0     0
    0     0     0  ...      0     0     0
       ...          ⋱          ...       
    0     0     0  ...      0     0     0
    0     0     0  ...      0     0     0
    0     0     0  ...      0     0     0
[torch.cuda.FloatTensor of size 1568x8 (GPU 0)]

I checked MarketplaceFeldkirch_Station4.h5 under /features/test_reduced/ folder and the labels field seems empty - dimension size is 0, and max dimension size is also 0.

In case there are relevant:

I've unzipped the point cloud files I downloaded from Semantic3D website, so there are .txt files in the data folder for each point cloud, rather than the zipped files.

I've placed .labels files into the same folder for the training files. There are no .labels files for either test category.

loicland, 2018-06-07:
Ok so that shouldn't be!

So just tomake sure, your /data folder contains 3 folders, /train /test_reduced and /test_full. Each of them contains the .txt files obtained when uncompressing the zip. The .labels files should be in the /data folder.

To try to pinpoint where it goes wrong, can you check the value of `has_labels`  and `labels ` at line 133 of /partition/partition.py?

DrFridolin, 2018-06-07:
I will let you know the values of has_labels and labels shortly, but just to make sure the data locations are right:

My /data folder contains those folders and each folder contains the corresponding .txt files. But the .labels files are in /data/train/ folder, next to the .txt files they belong. Is this wrong?

loicland, 2018-06-07:
my bad i typed too fast. what you did is correct.

DrFridolin, 2018-06-07:
`has_labels` is set to False for data under test_reduced and test_full, but it is set to True for data under train.

I think this is to be expected, since test data does not have corresponding .labels files in the same folder, but training data does.

Consequently, the contents of `labels` is empty for all the test data. Trying to see what `labels` contain for training data, it's taking a while to run on the debugger.

DrFridolin, 2018-06-07:
Here is a screenshot showing the first several rows of the labels variable.

The entire thing is too large and I thought it would be impractical to try to post the whole thing here.

[image]

loicland, 2018-06-07:
This is what it is supposed to look like yes. The first column is the number of points in the superpoint, the next one the number of unlabelled points, and the next 13 the number of points in each classes.

The feature file should be how you describe it, with labels for the training set and none (just [ ]) for test sets.

To see what is going on with label writing, put a stopping point line 346 of /partition/provider.py

DrFridolin, 2018-06-07:
I will do that, though my debugger seems to be able to write the labels fine so I'm sure whatever is going wrong otherwise, I won't be able to observe it in the debugger.

I am also trying to rerun the partitioning code with sudo access to see if it makes a difference.

There is something I don't understand though: If I download the pretrained model and try to do segmentation on test_reduced with that file, are these label fields significant?

The way I see it, the test data will have an empty labels field anyway, and the training data will not be used as I am using a pre-trained model. Am I missing something?

loicland, 2018-06-08:
Yep you are correct that the labels of the training set shouldn't impact the inference with the trained model.

Hopefully investigating this label writing problem will help us understanding what is going wrong.

DrFridolin, 2018-06-08:
Hi Loic,

As I suspected, the label writing problem disappeared when I made sure the code has sudo access. I can view the label fields in the .h5 files for the training data now. 

I am thinking that we no longer need to look into what's happening with labels at this point since they are working as intended - would you agree?

I am re-running the whole pipeline with proper sudo privileges to see if I can get some segmentation results. This will take a few hours since the partitioning has to be re-run. I will let you know what kinds of results I get as soon as I get them.

In the meantime, if you have any suggestions on what I can look at to make sure things are running smoothly, let me know. For instance - should I go back and see the contents of `label_mode` and `label_vec` ?

Many thanks for your help.

DrFridolin, 2018-06-11:
Hi Loic,

I just wanted to let you know that with proper sudo access, the .h5 files had the proper label fields and I was able to get prediction results without any issues.

I am actually not sure if label fields had anything to do with this, as I was loading the pre-trained model and testing it on reduced test set, which does not have labels to begin with. However I think it's possible that other things were going wrong without sudo access, which are now probably working correctly.

I have also started re-training the model on my end and the loss does not go to 0 after a mere 3 iterations like it used to. I'm guessing that the training is also working as intended.

Thank you very much for your time and effort and I'm sorry this turned out to be such a disappointing issue :) I will next try to run your code on my own data, and eventually train on my own data. As far as I can see you've already included instructions on how I can do that, so I'm hoping those will go smoothly as well.

If you have no further comments, I will close this thread now. Many thanks.

loicland, 2018-06-11:
Glad to hear it was fixed by a simple sudo!

Do not hesitate to ask for advice if you struggle to train your model. For starters, if you only have a small database I would advise you to use vv-convolution (vector-vector) and to decrease the size of the pointnets.

closed 2018-06-11 by loicland

chahatchawla001, 2018-12-12:
Hi, I am facing the same issue. Working in python virtual environment(and hence non-sudo access), Could you suggest some possible sources of error. Starting from scratch with sudo access is possible though, but need help in debugging where the exact source of error could be, since that would be more insightful in getting it fixed. 
Trying to avoid the process from start itself with sudo access since the partition process took 15 hours nearly on my system . Any suggestion in debugging or fastening the process of partition welcomed. 
Thanks!

chahatchawla001, 2018-12-12:
This needs to be re-opened, I tried the entire process with SUDO privileges at any of the step, but the labels being empty problem persists.Looks like source of error is somewhere else. @DrFridolin , did you change any thing(however subtle) in the way you proceeded ? Highly possible that something else you did, went unnoticed attributing the error to the sudo thing.

loicland, 2018-12-12:
Hi,

So just to be clear, the labels when loading your training data are all zero?
Can you post the first few lines of the label array before and after pruning?