loicland/superpoint_graph #39: Partitioning taking very long

amandaloh16, 2018-06-20:
Hello, 
Was wondering if people can provide me with how long the partition step takes. When I started using this code: it didn't take very long (minutes), but now it takes 2+ hours tops (don't know if this is a memory issue?). This is just with the base Semantic3D training.

 I also recently modified the training to take in an additional 1.8GB of additional data and now it takes more than 24 hours to partition. I lowered the batch to 1000000 but no resolve.

How long does partitioning take for you with the original data? With additional training data you may have added in addition to the Semantic3D one?

These are the parameters I'm using: 

--voxel_width 0.01 --reg_strength 0.8 --ver_batch 1000000

here is my RAM use in GB:
free -g
              total        used        free      shared  buff/cache   available
Mem:            125          24          52           0          48         100
Swap:           127           0         127

loicland, 2018-06-20:
Hi,

Thinks not normal, partitioning should process at least 100k points per second. This is most likely a memory issue. Try to increase the `voxel_width` parameter (we used 0.05 in the paper) after deleting (or renaming) your current `features` folder.

In my experience you won't lose too much precision and gain considerable computational acceleration. You can the upsample to the original data if you need to using `write_semantic3d.py`.

closed 2018-07-24 by loicland

amandaloh16, 2018-07-25:
Found part of the reason is the np.genfromtxt function can be very costly. If one uses pandas io operations you can save 11x. Was able to process 35 million points approx down from 186 seconds to 17 seconds. Maybe it doesn't matter as much if you have a better PC, but my PC definitely had major improvements with the partitioning code after this optimization.

loicland, 2018-07-27:
Hi,

did you try chosing an appropriate `--ver_batch` parameter? `np.genfromtxt` can be prohibitively slow when large files are read in one go, but behave better by when proceding by chunks.

Eitherway, I am not surprised Panda does better, I will look into it.

amandaloh16, 2018-07-27:
I kept the ver_batch parameter and experimenting with reducing down 5million to 1 million: didn't make much difference.

you can take a look at where the partition.py is spending time:

ncalls  tottime  percall  cumtime  percall filename:lineno(function)
     1174 16995.155   14.476 49715.980   42.348 /home/anaconda3/lib/python3.6/site-packages/numpy/lib/npyio.py:1489(genfromtxt)
123484177791 14802.648    0.000 15182.652    0.000 {built-in method builtins.next}
        1 5761.851 5761.851 63018.917 63018.917 partition/partition.py:6(<module>)
      114 4105.269   36.011 5141.390   45.100 partition/graphs.py:48(compute_sp_graph)
17814865073 3690.404    0.000 4270.255    0.000 /home/anaconda3/lib/python3.6/site-packages/numpy/lib/_iotools.py:704(_loose_call)
4379806521 2214.876    0.000 4592.733    0.000 /home/anaconda3/lib/python3.6/site-packages/numpy/lib/_iotools.py:232(_delimited_splitter)
8759616722 1754.061    0.000 1754.061    0.000 {method 'split' of 'str' objects}
4379806521 1668.881    0.000 6971.995    0.000 /home/anaconda3/lib/python3.6/site-packages/numpy/lib/_iotools.py:262(__call__)
  9627636 1193.564    0.000 1193.564    0.000 {built-in method numpy.core.multiarray.array}
      114  999.242    8.765  999.769    8.770 {method 'query' of 'sklearn.neighbors.kd_tree.BinaryTree' objects}