loicland/superpoint_graph #41: Can I run you trained networks on a computer without GPU?

yanyu003, 2018-07-02:
Hi!
I am trying to run your trained network with the semantic3D reduced-dataset on my computer. But, I don't have GPU on my computer? I use the command: 
`_python3.5 learning/main.py --dataset sema3d --SEMA3D_PATH /home/us-wrr/code/superpoint_graph-release/sema3d --db_test_name testred --db_train_name trainval --epochs -1 --lr_steps '[350, 400, 450]' --test_nth_epoch 100 --model_config 'gru_10,f_8' --ptn_nfeat_stn 11 --nworkers 2 --odir "results/sema3d/trainval_best" --resume RESUME_
`
and have the error:
`loading checkpoint 'results/sema3d/trainval_best/model.pth.tar'
THCudaCheck FAIL file=torch/csrc/cuda/Module.cpp line=51 error=35 : CUDA driver version is insufficient for CUDA runtime version
Traceback (most recent call last):
  File "learning/main.py", line 387, in <module>
    main()
  File "learning/main.py", line 144, in main
    model, optimizer, stats = resume(args, dbinfo)
  File "learning/main.py", line 328, in resume
    checkpoint = torch.load(args.resume)
  File "/home/us-wrr/.local/lib/python3.5/site-packages/torch/serialization.py", line 303, in load
    return _load(f, map_location, pickle_module)
  File "/home/us-wrr/.local/lib/python3.5/site-packages/torch/serialization.py", line 469, in _load
    result = unpickler.load()
  File "/home/us-wrr/.local/lib/python3.5/site-packages/torch/serialization.py", line 437, in persistent_load
    data_type(size), location)
  File "/home/us-wrr/.local/lib/python3.5/site-packages/torch/serialization.py", line 88, in default_restore_location
    result = fn(storage, location)
  File "/home/us-wrr/.local/lib/python3.5/site-packages/torch/serialization.py", line 70, in _cuda_deserialize
    return obj.cuda(device)
  File "/home/us-wrr/.local/lib/python3.5/site-packages/torch/_utils.py", line 68, in _cuda
    with torch.cuda.device(device):
  File "/home/us-wrr/.local/lib/python3.5/site-packages/torch/cuda/__init__.py", line 225, in __enter__
    self.prev_idx = torch._C._cuda_getDevice()
RuntimeError: cuda runtime error (35) : CUDA driver version is insufficient for CUDA runtime version at torch/csrc/cuda/Module.cpp:51
`
How can I run your trained network  on my computer? Or, I need to train the network again?

loicland, 2018-07-02:
Hi,

first you need to set the option `--cuda` to 0. But yes, the trained weights are only compatible with cuda, and you will need to retrain the model from scratch. It takes about 3 hours on my 1080. I haven't tried without CPUs, but it might take a while.

closed 2018-07-17 by loicland

bermanmaxim, 2018-07-20:
There are ways to load GPU weights to CPU only (changing device) with torch.load, have a look at:
https://discuss.pytorch.org/t/loading-weights-for-cpu-model-while-trained-on-gpu/1032

amandaloh16, 2018-07-27:
If i already have a model.pth.tar file, I don't need a GPU computer to semantically segment new test data correct? What is the correct command line arg to do that? Is this correct? Say I have a folder of new data to label in NewClouds

CUDA_VISIBLE_DEVICES=0 python learning/main.py --cuda 0 --dataset sema3d --SEMA3D_PATH $SEMA3D_DIR --db_test_name NewClouds --db_train_name train \
--epochs -1 --lr_steps '[350, 400, 450]' --test_nth_epoch 100 --model_config 'gru_10,f_8' --ptn_nfeat_stn 11 \
--nworkers 2 --odir "results/sema3d/trainval_best" --resume <where model.pth.tar lives>

loicland, 2018-11-07:
Hi,

sorry a bit late to the party, but the link sent by @bermanmaxim works.

the simplest way would be to add:

```
if not args.cuda:
        model = model.cpu()
 ```

in the function `resume`

mys007, 2018-11-07:
Hi, should also work (and allows you to load the model even if you have no cuda at all:
```
checkpoint = torch.load(args.resume, map_location=None if args.cuda else 'cpu')
```