loicland/superpoint_graph #47: Partitioning Semantic 3d dataset - Performance

SabaMehmood15, 2018-08-05:
I am currently executing partition step for Semantic 3d dataset but it is taking a lot of time. How much time approximately does this script takes to partition all test and training point clouds of Semantic 3d dataset?
The script has been executing for 2 days approx and still it is partitioning first pointcloud (sg28_Station2_rgb_intensity-reduced of Reduced Test set) . My machine has 16 GB Ram and the code is utilizing swap memory too.  I have increased voxel_width to 0.5 and ver_batch decreased to 500. Am I doing something wrong?
Also can you provide me the partitioned output files i.e. .h5 format for Semantic 3d dataset?

amandaloh16, 2018-08-06:
take a look at this thread...I had same issue: https://github.com/loicland/superpoint_graph/issues/39

SabaMehmood15, 2018-08-07:
Yes 24 GB RAM is recommended. Did you manage to solve the issue? Actually I am more interested in partitioned files for Semantic 3d dataset if someone can provide me?

loicland, 2018-08-07:
Hi,

2 days is definitively not right. Even the largest point clouds takes a couple dozen minutes tops, as just tested on my computer with 16GB as well with `voxel_width` 0.05 and `ver_batch 5000000`. Would you be able to tell me which call takes such a long time? Is it `read_semantic3d_format`? Or the partition itself during the call to `cut-pursuit`?

Can you run it in debug mode and try to see which part causes you to run out of ram?

The output in the terminal must be like this:
```
=================
   train/
=================
1 / 15---> sg28_station2
    creating the feature file...
=========================
======== pruning ========
=========================
Voxelization into 3432 x 2099 x 47 grid
Reduced from 5000000 to 616773 points (12.33%)
=========================
======== pruning ========
=========================
Voxelization into 320 x 240 x 43 grid
Reduced from 5000000 to 52540 points (1.05%)
=========================
======== pruning ========
=========================
Voxelization into 160 x 160 x 42 grid
Reduced from 5000000 to 10446 points (0.2%)
...
```

How long does each pruning block takes to compute?

SabaMehmood15, 2018-08-08:
Hi Loic Landrieu,

For the training dataset the script is working fine, this is the problem with test_reduces set. It got stuck on first pointcloud i.e. sg28_Station2_rgb_intensity-reduced. I tried to execute it again with the parameters you provided but still got the same problem. I am attaching screen shot for the command I executed, it was taken after 3 hours of wait.
[image]

loicland, 2018-08-08:
The problem seems to be during the cut pursuit call indeed. Does your RAM gets saturated during this call? Cut pursuit can be very slow on swap memory (but 3 hours for zero iteration seems extreme).

Can you try on another smaller file (by just changing the directory of sg28_Station2). If it turns out that it was indeed the problem, you will need to increase the voxel_size (to 0.1 for example).

If this still takes a very long time, please paste the terminal as you did here as well.

I gather that you have computed all the `features` files?

SabaMehmood15, 2018-08-14:
Thank you, increasing voxel_size to 0.1 worked for me.

closed 2018-08-24 by loicland