loicland/superpoint_graph #50: Inference error in  line 336, in resume     model.load_state_dict(checkpoint['state_dict']

chowkamlee81, 2018-08-21:
Hi,

I wanted to do inference on semantic_3d dataset. I downloaded the pretrained model from web and remaed to model.pth.tar. 
Kindly help, But it landed error throwing exception:

Traceback (most recent call last):
  File "/usr/share/pycharm/helpers/pydev/pydevd.py", line 1599, in <module>
    globals = debugger.run(setup['file'], None, None, is_module)
  File "/usr/share/pycharm/helpers/pydev/pydevd.py", line 1026, in run
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "/usr/share/pycharm/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/home/administrator/Documents.orig/Kumar_FileSystem/Server_Code/sematic3d/superpoint_graph-release/learning/main.py", line 387, in <module>
    main()
  File "/home/administrator/Documents.orig/Kumar_FileSystem/Server_Code/sematic3d/superpoint_graph-release/learning/main.py", line 144, in main
    model, optimizer, stats = resume(args, dbinfo)
  File "/home/administrator/Documents.orig/Kumar_FileSystem/Server_Code/sematic3d/superpoint_graph-release/learning/main.py", line 336, in resume
    model.load_state_dict(checkpoint['state_dict'])
  File "/home/administrator/anaconda3/lib/python3.6/site-packages/torch/nn/modules/module.py", line 719, in load_state_dict
    self.__class__.__name__, "\n\t".join(error_msgs)))
RuntimeError: Error(s) in loading state_dict for Module:
	Unexpected running stats buffer(s) "ecc.0._cell.ini.running_mean" and "ecc.0._cell.ini.running_var" for InstanceNorm1d with track_running_stats=False. If state_dict is a checkpoint saved before 0.4.0, this may be expected because InstanceNorm1d does not track running stats by default since 0.4.0. Please remove these keys from state_dict. If the running stats are actually needed, instead set track_running_stats=True in InstanceNorm1d to enable them. See the documentation of InstanceNorm1d for details.
	Unexpected running stats buffer(s) "ecc.0._cell.inh.running_mean" and "ecc.0._cell.inh.running_var" for InstanceNorm1d with track_running_stats=False. If state_dict is a checkpoint saved before 0.4.0, this may be expected because InstanceNorm1d does not track running stats by default since 0.4.0. Please remove these keys from state_dict. If the running stats are actually needed, instead set track_running_stats=True in InstanceNorm1d to enable them. See the documentation of InstanceNorm1d for details.

Process finished with exit code 1

loicland, 2018-08-21:
Hi,

SPG is still only compatible with pyorch 3.1. We are working on it and will upgrade soon. In the meantime, I advise you to create a virtual env with 3.1.

Loic

chowkamlee81, 2018-08-21:
**Kindly help**
Downgraded to version pytorch3.1 rather than 4.0. But got an error below.
 0%|                                                                        | 0/14 [00:00<?, ?it/s]Traceback (most recent call last):
  File "learning/main.py", line 387, in <module>
    main()
  File "learning/main.py", line 311, in main
    acc_test, oacc_test, avg_iou_test, per_class_iou_test, predictions_test, avg_acc_test, confusion_matrix = eval_final()
  File "learning/main.py", line 254, in eval_final
    outputs = model.ecc(embeddings)
  File "/home/administrator/anaconda3/lib/python3.6/site-packages/torch/nn/modules/module.py", line 357, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/administrator/Documents.orig/FileSystem/Server_Code/sematic3d/superpoint_graph-release/learning/graphnet.py", line 97, in forward
    input = module(input)
  File "/home/administrator/anaconda3/lib/python3.6/site-packages/torch/nn/modules/module.py", line 357, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/administrator/Documents.orig/FileSystem/Server_Code/sematic3d/superpoint_graph-release/learning/modules.py", line 54, in forward
    input = ecc.GraphConvFunction(nc, nc, idxn, idxe, degs, degs_gpu, self._edge_mem_limit)(hx, weights)
  File "/home/administrator/Documents.orig/FileSystem/Server_Code/sematic3d/superpoint_graph-release/learning/ecc/GraphConvModule.py", line 67, in forward
    cuda_kernels.conv_aggregate_fw(output.narrow(0,startd,numd), products.view(-1,self._out_channels), self._degs_gpu.narrow(0,startd,numd))
  File "/home/administrator/Documents.orig/FileSystem/Server_Code/sematic3d/superpoint_graph-release/learning/ecc/cuda_kernels.py", line 120, in conv_aggregate_fw
    function, stream = get_kernel_func('conv_aggregate_fw_kernel_v2', conv_aggregate_fw_kernel_v2(), get_dtype(src))
  File "/home/administrator/Documents.orig/FileSystem/Server_Code/sematic3d/superpoint_graph-release/learning/ecc/cuda_kernels.py", line 33, in get_kernel_func
    prog = Program(ksrc.encode('utf-8'), (kname+dtype+'.cu').encode('utf-8'))
  File "/home/administrator/anaconda3/lib/python3.6/site-packages/pynvrtc/compiler.py", line 52, in __init__
    include_names)
  File "/home/administrator/anaconda3/lib/python3.6/site-packages/pynvrtc/interface.py", line 200, in nvrtcCreateProgram
    c_char_p(encode_str(src)), c_char_p(encode_str(name)),
  File "/home/administrator/anaconda3/lib/python3.6/site-packages/pynvrtc/interface.py", line 54, in encode_str
    return s.encode("utf-8")
AttributeError: 'bytes' object has no attribute 'encode'
Exception ignored in: <bound method Program.__del__ of <pynvrtc.compiler.Program object at 0x7f828338c4a8>>
Traceback (most recent call last):
  File "/home/administrator/anaconda3/lib/python3.6/site-packages/pynvrtc/compiler.py", line 56, in __del__
    self._interface.nvrtcDestroyProgram(self._program)
AttributeError: 'Program' object has no attribute '_program'
^Z
[6]+  Stopped                 CUDA_VISIBLE_DEVICES=0,1 python learning/main.py --dataset sema3d --SEMA3D_PATH /home/administrator/Documents.orig/FileSystem/Server_Code/sematic3d --db_test_name testfull --db_train_name trainval --epochs -1 --lr_steps '[350, 400, 450]' --test_nth_epoch 100 --model_config 'gru_10,f_8' --ptn_nfeat_stn 11 --nworkers 2 --odir "/home/administrator/Documents.orig/FileSystem/results/sema3d/trainval_best" --resume RESUME

chowkamlee81, 2018-08-21:
solved by using  
replacing line 3̶5̶ 33 of /learning/ecc/cuda_kernels.py by

prog = Program(ksrc, (kname+dtype+'.cu'))

Closing the issue

closed 2018-08-21 by chowkamlee81

mys007, 2018-08-30:
This was caused by recently merged https://github.com/NVIDIA/pynvrtc/pull/2 and fixed here today - https://github.com/loicland/superpoint_graph/commit/44bcbe8921d01988d57cb05953b7b521bf0b7cf0