loicland/superpoint_graph #53: prediction_testred.h5 not created

chowkamlee81, 2018-08-21:
after executing the below commands, 
python partition/visualize.py --dataset sema3d --ROOT_PATH /home/administrator/Documents.orig/Kumar_FileSystem/Server_Code/sematic3d --res_file '/home/administrator/Documents.orig/Kumar_FileSystem/results/sema3d/trainval_best/prediction_testred' --file_path 'test_reduced/MarketplaceFeldkirch_Station4' --output_type ifprs
OR
CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset sema3d --SEMA3D_PATH /home/administrator/Documents.orig/Kumar_FileSystem/Server_Code/sematic3d --db_test_name testred --db_train_name trainval --epochs 500 --test_nth_epoch 100 --model_config 'gru_10,f_8' --ptn_nfeat_stn 11 --nworkers 2 --odir "/home/administrator/Documents.orig/Kumar_FileSystem/results/sema3d/trainval_best"

got the error below

./results/sema3d/trainval_best/predictions_testred.h5
Traceback (most recent call last):
  File "partition/write_Semantic3d.py", line 54, in <module>
    res_file = h5py.File(res_folder + 'predictions_' + args.db_test_name + '.h5', 'r')   
  File "/home/administrator/anaconda3/lib/python3.6/site-packages/h5py/_hl/files.py", line 269, in __init__
    fid = make_fid(name, mode, userblock_size, fapl, swmr=swmr)
  File "/home/administrator/anaconda3/lib/python3.6/site-packages/h5py/_hl/files.py", line 99, in make_fid
    fid = h5f.open(name, flags, fapl=fapl)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5f.pyx", line 78, in h5py.h5f.open
OSError: Unable to open file (unable to open file: name = './results/sema3d/trainval_best/predictions_testred.h5', errno = 2, error message = 'No such file or directory', flags = 0, o_flags = 0)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "partition/write_Semantic3d.py", line 56, in <module>
    raise ValueError("%s do not exists" % res_file) 
NameError: name 'res_file' is not defined

loicland, 2018-08-22:
Does the file `./results/sema3d/trainval_best/predictions_testred.h5` exist in your spg folder? It should be created after you run inference on the reduced test. Relaunch `learning/main.py` on the reduced test with `--epochs -1` to make sure.

renhaofan, 2018-09-15:
@loicland I met the simliar problem..
```
$ python partition/visualize.py --dataset sema3d --ROOT_PATH /home/renhaofan/dataset/semantic3d --res_file 'results/sema3d/trainval_best/prediction_testred' --file_path 'test_reduced/MarketplaceFeldkirch_Station4' --output_type ifprs

/home/renhaofan/anaconda3/lib/python3.6/site-packages/h5py/__init__.py:36: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
Traceback (most recent call last):
  File "partition/visualize.py", line 64, in <module>
    raise ValueError("%s does not exist and is needed to output the result ply" % res_file) 
ValueError: results/sema3d/trainval_best/prediction_testred.h5 does not exist and is needed to output the result ply

```
but in my ./results/sema3d/trainval_best/ `predictions_testred.h5` exits.
```
-rw-rw-r-- 1 renhaofan renhaofan     309 Sep 15 16:53 cmdline.txt
-rw-rw-r-- 1 renhaofan renhaofan 2612591 Sep 15 17:41 model.pth.tar
-rw-rw-r-- 1 renhaofan renhaofan     640 Sep 15 17:44 pointwise_cm.npy
-rw-rw-r-- 1 renhaofan renhaofan  454504 Sep 15 10:52 predictions_testfull.h5
-rw-rw-r-- 1 renhaofan renhaofan  311128 Sep 15 17:44 predictions_testred.h5
-rw-rw-r-- 1 renhaofan renhaofan     258 Sep 15 10:52 scores_testfull.txt
-rw-rw-r-- 1 renhaofan renhaofan     256 Sep 15 17:44 scores_testred.txt
-rw-rw-r-- 1 renhaofan renhaofan   70486 Sep 15 17:41 trainlog.txt

```
And I have tried to . `Relaunch learning/main.py `, but it useless. 
About the `--output_type ifprs`, I found the error above is caused by `--output_type r`, any suggestions?