loicland/superpoint_graph #54: AttributeError: 'bytes' object has no attribute 'encode'

renhaofan, 2018-08-29:
I'm on ubuntu16.04 anaconda python3.6
after executing the below commands:
`CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset sema3d --SEMA3D_PATH /home/sdc/dataset/semantic3d --db_test_name testred --db_train_name trainval --epochs 500 --lr_steps '[350, 400, 450]' --test_nth_epoch 100 --model_config 'gru_10,f_8' --ptn_nfeat_stn 11 --nworkers 2 --odir "results/sema3d/trainval_best"`
got the error below:

Epoch 0/500 (results/sema3d/trainval_best):
  0%|                                                                         | 0/7 [00:00<?, ?it/s]Traceback (most recent call last):
  File "learning/main.py", line 387, in <module>
    main()
  File "learning/main.py", line 286, in main
    acc, loss, oacc, avg_iou = train()
  File "learning/main.py", line 184, in train
    outputs = model.ecc(embeddings)
  File "/home/renhaofan/anaconda3/lib/python3.6/site-packages/torch/nn/modules/module.py", line 325, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/renhaofan/superpoint_graph/learning/graphnet.py", line 97, in forward
    input = module(input)
  File "/home/renhaofan/anaconda3/lib/python3.6/site-packages/torch/nn/modules/module.py", line 325, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/renhaofan/superpoint_graph/learning/modules.py", line 54, in forward
    input = ecc.GraphConvFunction(nc, nc, idxn, idxe, degs, degs_gpu, self._edge_mem_limit)(hx, weights)
  File "/home/renhaofan/superpoint_graph/learning/ecc/GraphConvModule.py", line 67, in forward
    cuda_kernels.conv_aggregate_fw(output.narrow(0,startd,numd), products.view(-1,self._out_channels), self._degs_gpu.narrow(0,startd,numd))
  File "/home/renhaofan/superpoint_graph/learning/ecc/cuda_kernels.py", line 120, in conv_aggregate_fw
    function, stream = get_kernel_func('conv_aggregate_fw_kernel_v2', conv_aggregate_fw_kernel_v2(), get_dtype(src))
  File "/home/renhaofan/superpoint_graph/learning/ecc/cuda_kernels.py", line 33, in get_kernel_func
    prog = Program(ksrc.encode('utf-8'), (kname+dtype+'.cu').encode('utf-8'))
  File "/home/renhaofan/anaconda3/lib/python3.6/site-packages/pynvrtc/compiler.py", line 52, in __init__
    include_names)
  File "/home/renhaofan/anaconda3/lib/python3.6/site-packages/pynvrtc/interface.py", line 200, in nvrtcCreateProgram
    c_char_p(encode_str(src)), c_char_p(encode_str(name)),
  File "/home/renhaofan/anaconda3/lib/python3.6/site-packages/pynvrtc/interface.py", line 54, in encode_str
    return s.encode("utf-8")
`AttributeError: 'bytes' object has no attribute 'encode'`
Exception ignored in: <bound method Program.__del__ of <pynvrtc.compiler.Program object at 0x7f8feb1ef240>>
Traceback (most recent call last):
  File "/home/renhaofan/anaconda3/lib/python3.6/site-packages/pynvrtc/compiler.py", line 56, in __del__
    self._interface.nvrtcDestroyProgram(self._program)
`AttributeError: 'Program' object has no attribute '_program'`

For bo
For the first error, I knowed it need to encode and decode between `string` and `bytes`in python3, so I vim the interface.py
```
 51  def encode_str(s):
 52      if is_python2:
 53          return s
 54      return s.encode("utf-8")
```
I think the code snippet is right. For the two errors, I have searched, but all of them I think are helpless. Could tell me waht's the py version you utilize? or maybe paste the your `compiler.py` and `interface.py` please

renhaofan, 2018-08-30:
I should add that for the second error, From [here](https://github.com/NVIDIA/FastPhotoStyle/issues/10)  I learned that it's for python2, but your python version is 3.6. So I'm confused that how you can run the code

loicland, 2018-08-30:
Hi,

this seems to be an error a few other people are encountering. It was solved in [https://github.com/loicland/superpoint_graph/issues/43](this issue) and [https://github.com/loicland/superpoint_graph/issues/49](this one).

Let me know if it doesn't work for you.

mys007, 2018-08-30:
This was caused by recently merged https://github.com/NVIDIA/pynvrtc/pull/2 and fixed here today - https://github.com/loicland/superpoint_graph/commit/44bcbe8921d01988d57cb05953b7b521bf0b7cf0

closed 2018-08-30 by mys007

renhaofan, 2018-08-31:
@loicland thankyou, by your guide, I have successfully solved the issue. @mys007  that's it