loicland/superpoint_graph #58: Any possibility to train network of semantic 3D on multi-gpu?

staochun, 2018-10-10:
I have tried to train s3d network on a single GPU according to your instruction. However, there is usually an out-of-memory error on CUDA during the training epochs so that the training cannot experience enough epochs. So I would like to ask for some improvement advice for upgrading to multi-GPU version. I have tried several ways but no one succeeds. Thanks for your kind help very much.

loicland, 2018-10-11:
Hi,

I don't really know anyhting about multiple GPU setup, but I am surprised that you are running out of memory. We were able to train the model on a GPU with only 6BG of memory thanks to the momory monger.

Are you using the recommended reg_strength?
What is your setup?
Have you tries increasing the voxel width to decrease the cloud size?

HenrryBryant, 2019-10-03:
hi, sure you can do it. just compose ptn and ecc into one model and wrap this model with nn.DataParallel()