loicland/superpoint_graph #61: python partition/partition.py --dataset s3dis --ROOT_PATH dataset --voxel_width 0.03 --reg_strength 0.03

ytgcljj, 2018-10-25:
python partition/partition.py --dataset s3dis --ROOT_PATH dataset --voxel_width 0.03 --reg_strength 0.03
Traceback (most recent call last):
  File "partition/partition.py", line 15, in <module>
    import libcp
ImportError: ./partition/cut-pursuit/src/libcp.so: undefined symbol: _ZN5boost6python6detail11init_moduleER11PyModuleDefPFvvE

Inventor0, 2018-11-14:
I meet the same question 
May I ask what may have caused it?

ubuntu16.04
python3.6
Boost version: 1.68.0

Traceback (most recent call last):
  File "partition/partition.py", line 15, in <module>
    import libcp
ImportError: ./partition/cut-pursuit/src/libcp.so: undefined symbol: _ZN5boost6python6detail11init_moduleER11PyModuleDefPFvvE

loicland, 2018-11-14:
Hi,

the problem is a linking problem with libboost. can you try @yxchng's solution in this thread (https://github.com/BVLC/caffe/issues/5810)?

Inventor0, 2018-11-17:
I want to know
python partition/partition.py --dataset s3dis --ROOT_PATH $S3DIR_DIR --voxel_width 0.03 --reg_strength 0.03
 use python2 or python3 ？

loicland, 2018-11-18:
I haven't tested it with python 2, but I it should work provided you were able to build the .libraries (plyclib and cplib.so), and have been able to succesfully link boost to python 2. 

I have tried it extensively with 3.5 and 3.6 however.

zeroAska, 2019-04-25:
I have a similar issue with conda when I ran `python3.7 partition/partition.py --dataset sema3d --ROOT_PATH semantic3d_data/ --voxel_width 0.05 --reg_strength 0.8 --ver_batch 5000000`:

Traceback (most recent call last):
  File "partition/partition.py", line 15, in <module>
    import libcp
ImportError: ./partition/cut-pursuit/src/libcp.so: undefined symbol: _ZTIN5boost6python7objects21py_function_impl_baseE

If I type `locate anaconda3/lib/libpython`, I got:
/home/v9999/anaconda3/lib/libpython3.7m.a
/home/v9999/anaconda3/lib/libpython3.7m.so
/home/v9999/anaconda3/lib/libpython3.7m.so.1
/home/v9999/anaconda3/lib/libpython3.7m.so.1.0

loicland, 2019-04-26:
Hi,

it seems that there was an issue in the building of the wrappers. Can you post the `make ...` comamnds you used to build them?

zeroAska, 2019-04-27:
Thanks for the help!

CMakelists: 
CMAKE_MINIMUM_REQUIRED(VERSION 3.5)

PROJECT(CUT_PURSUIT_SEG)

set (CMAKE_CXX_STANDARD 11)

set(CMAKE_MODULE_PATH ${CMAKE_CURRENT_SOURCE_DIR}/CMakeModules ${CMAKE_MODULE_PATH})

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall -std=c++11 -fopenmp -O3")
##############################
### Find required packages ###
##############################

include_directories(../include)

find_package(Boost 1.67 COMPONENTS graph numpy python REQUIRED) #system filesystem thread serialization
include_directories(${BOOST_INCLUDE_DIRS})
link_directories(${BOOST_LIBRARYDIR})
message("Boost include " ${BOOST_INCLUDEDIR})
message("Boost lib " ${BOOST_LIBRARYDIR})

find_package(PythonLibs)
SET(PYTHON_LIBRARIES /var/data/rvm/conda_env/lib/libpython3.7m.so)
SET(PYTHON_INCLUDE_DIRS /var/data/rvm/conda_env/include/python3.7m/)
SET(NUMPY_INCLUDE_DIRS /var/data/rvm/conda_env/lib/python3.7/site-packages/numpy/core/include/)
message("PYTHON LIBRARIES ARE " ${PYTHON_LIBRARIES})
INCLUDE_DIRECTORIES(${PYTHON_INCLUDE_DIRS} ${NUMPY_INCLUDE_DIRS} )
LINK_DIRECTORIES(${PYTHON_LIBRARY_DIRS})

file(GLOB CP_HEADERS include/*.h)

set(CMAKE_LD_FLAG "${CMAKE_LD_FLAGS} -shared -Wl -fPIC --export-dynamic -O3 -Wall -std=c++11")

add_library(cp SHARED cutpursuit.cpp ${CP_HEADERS})
target_link_libraries(cp
	${BOOST_LIBRARIEDIR}
    ${PYTHON_LIBRARIES}
    )

cmake command: 
cmake . -DPYTHON_LIBRARY=$CONDA_PREFIX/lib/libpython3.7m.so -DPYTHON_INCLUDE_DIR=$CONDA_PREFIX/include/python3.7m/ -DBOOST_LIBRARYDIR=$CONDA_PREFIX/lib/ -DEIGEN3_INCLUDE_DIR=$CONDA_PREFIX/include/eigen3

zeroAska, 2019-04-27:
I fixed this! My way of finding boost packages in CMakeList.txt is not correct.

loicland, 2019-04-27:
Great! It seems that the virtual environment had the `find_package` function struggling. What did you set your `CONDAENV` to?

zeroAska, 2019-04-27:
Thanks loiland! I use the path for my conda environment, which is  /var/data/rvm/conda_env/.  The actual problem is that the boost importing in cmake is case sensitive. For example, Boost_LIBRARY_DIR and BOOST_LIBRARYDIR are different...

closed 2019-04-29 by loicland

CharlieLeee, 2019-10-07:
Sorry to open this issue again, I happened to run into this error when trying to test on the Semantic3D dataset. I'm wondering what should the path of Boost_LIBRARIES be?

loicland, 2019-10-07:
Hi,

can you post the exact error that you have, along with what you set your `CONDA_ENV` to and your python version?

CharlieLeee, 2019-10-08:
HI loicland, here's my CONDA_ENV
CONDAENV=/home/undergraduate/miniconda3/envs/sp

python version is 3.7

the command I used to build cut-pursuit:
`cmake .. -DPYTHON_LIBRARY=/home/undergraduate/miniconda3/envs/sp/lib/libpython3.7m.so -DPYTHON_INCLUDE_DIR=/home/undergraduate/miniconda3/envs/sp/include/python3.7m -DBOOST_INCLUDEDIR=/home/undergraduate/miniconda3/envs/sp/include -DEIGEN3_INCLUDE_DIR=/home/undergraduate/miniconda3/envs/sp/include/eigen3

`
the CMakeLists.txt:
`
##############################
### Find required packages ###
##############################
find_package(PythonLibs)
find_package(PythonInterp)
#include(FindNumPy)
include_directories(../include)

#find_package(Boost 1.65.0 COMPONENTS graph REQUIRED PATHS /home/undergraduate/miniconda2/envs/spg/lib) #system filesystem thread serialization
#if (${Boost_MINOR_VERSION} LESS 67 )
#    find_package(Boost 1.65.0 COMPONENTS numpy${PYTHON_VERSION_MAJOR} REQUIRED PATHS /home/undergraduate/miniconda2/envs/spg/lib) #system filesystem thread serialization
#else()
#    set(PYTHONVERSION ${PYTHON_VERSION_MAJOR}${PYTHON_VERSION_MINOR})
#    find_package(Boost 1.67.0 COMPONENTS numpy${PYTHONVERSION} REQUIRED PATHS /home/undergraduate/miniconda2/envs/spg/lib)
#endif()

set(Boost_INCLUDE_DIRS /home/undergraduate/miniconda3/envs/sp/include/boost)
set(Boost_LIBRARY_DIRS /home/undergraduate/miniconda3/envs/sp/lib)

include_directories(${Boost_INCLUDE_DIRS})
link_directories(${Boost_LIBRARY_DIRS})

message("PYTHON LIBRARIES ARE " ${PYTHON_LIBRARIES})
INCLUDE_DIRECTORIES(${PYTHON_INCLUDE_DIRS} ${PYTHON_NUMPY_INCLUDE_DIR})
LINK_DIRECTORIES(${PYTHON_LIBRARY_DIRS})

file(GLOB CP_HEADERS include/*.h)

set(CMAKE_LD_FLAG "${CMAKE_LD_FLAGS} -shared -Wl -fPIC --export-dynamic -o -O3 -Wall -std=c++11")
# message(${Boost_LIBRARIES})
add_library(cp SHARED cutpursuit.cpp ${CP_HEADERS})
target_link_libraries(cp
    ${Boost_LIBRARIES}
    ${PYTHON_LIBRARIES})
`

the command I used to run semantic3d:
python partition/partition.py --dataset sema3d --ROOT_PATH /home/undergraduate/ChengkunLi/Sem3 --voxel_width 0.05 --reg_strength 0.8 --ver_batch 5000000

and error I got:
Traceback (most recent call last):
  File "partition/partition.py", line 15, in <module>
    import libcp
ImportError: ./partition/cut-pursuit/build/src/libcp.so: undefined symbol: _ZTIN5boost6python7objects21py_function_impl_baseE

loicland, 2019-10-08:
Mmh seems good to me. 

have you tried @zeroAska's solution of trying different capitalization of boost/BOOST in the cmake / the command line?

Also do you have multiple versions of boost installed on your computer? Potentialy one for python 2.X and one for Python 3.X? Try:

```
sudo find / -iname 'libboost_python*.so'
```

Can you post the results of `conda list` in your virtual environment?

CharlieLeee, 2019-10-09:
Hi loicland, thanks for the advice, I'll try this out as soon as I get access to the server 

Also, I would like to know what the `${Boost_LIBRARIES}` refer to in the CMakeLists.txt since I have commented `message(${Boost_LIBRARIES})` out to prevent cmake error. And I set the  `Boost_INCLUDE_DIRS` and `Boost_LIBRARY_DIRS` to my miniconda environment paths where boost lib and include are:
`set(Boost_INCLUDE_DIRS /home/undergraduate/miniconda3/envs/sp/include/boost)`
`set(Boost_LIBRARY_DIRS /home/undergraduate/miniconda3/envs/sp/lib)`
 
I don't know if the commented out `${Boost_LIBRARIES}` in the cmakelist file was the cause of this error? 

CMakeList.txt:
`##############################`
`### Find required packages ###`
`##############################`
`find_package(PythonLibs)`
`find_package(PythonInterp)`
`#include(FindNumPy)`
`include_directories(../include)`

`#find_package(Boost 1.65.0 COMPONENTS graph REQUIRED PATHS` `/home/undergraduate/miniconda2/envs/spg/lib) #system filesystem thread serialization`
`#if (${Boost_MINOR_VERSION} LESS 67 )`
`#    find_package(Boost 1.65.0 COMPONENTS numpy${PYTHON_VERSION_MAJOR} REQUIRED PATHS /home/undergraduate/miniconda2/envs/spg/lib) #system filesystem thread serialization`
`#else()`
`#    set(PYTHONVERSION ${PYTHON_VERSION_MAJOR}${PYTHON_VERSION_MINOR})`
`#    find_package(Boost 1.67.0 COMPONENTS numpy${PYTHONVERSION} REQUIRED PATHS /home/undergraduate/miniconda2/envs/spg/lib)`
`#endif()`

`set(Boost_INCLUDE_DIRS /home/undergraduate/miniconda3/envs/sp/include/boost)`
`set(Boost_LIBRARY_DIRS /home/undergraduate/miniconda3/envs/sp/lib)`

`include_directories(${Boost_INCLUDE_DIRS})`
`link_directories(${Boost_LIBRARY_DIRS})`

`message("PYTHON LIBRARIES ARE " ${PYTHON_LIBRARIES})`
`INCLUDE_DIRECTORIES(${PYTHON_INCLUDE_DIRS} ${PYTHON_NUMPY_INCLUDE_DIR})`
`LINK_DIRECTORIES(${PYTHON_LIBRARY_DIRS})`

`file(GLOB CP_HEADERS include/*.h)`

`set(CMAKE_LD_FLAG "${CMAKE_LD_FLAGS} -shared -Wl -fPIC --export-dynamic -o -O3 -Wall -std=c++11")`
`# message(${Boost_LIBRARIES})`
`add_library(cp SHARED cutpursuit.cpp ${CP_HEADERS})`
`target_link_libraries(cp
    ${Boost_LIBRARIES}
    ${PYTHON_LIBRARIES})
`

CharlieLeee, 2019-10-15:
@loicland the output of conda list is:
```
# packages in environment at /home/undergraduate/miniconda3/envs/sp:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main    defaults
boost                     1.67.0                   py37_4    anaconda
bzip2                     1.0.8                h7b6447c_0    anaconda
ca-certificates           2019.8.28                     0    anaconda
cairo                     1.14.12              h8948797_3    anaconda
certifi                   2019.9.11                py37_0    anaconda
cffi                      1.12.3           py37h2e261b9_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
chardet                   3.0.4                    pypi_0    pypi
cudatoolkit               10.0.130                      0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
eigen                     3.3.7                hfd86e86_0    defaults
eigen3                    3.3.7                         0    omnia
fastrlock                 0.4             py37hf484d3e_1000    conda-forge
fontconfig                2.13.0               h9420a91_0    anaconda
freetype                  2.10.0               he983fc9_1    conda-forge
future                    0.17.1                   py37_0    anaconda
gettext                   0.19.8.1          hc5be6a0_1002    conda-forge
glib                      2.56.2               hd408876_0    anaconda
gmp                       6.1.2             hf484d3e_1000    conda-forge
h5py                      2.10.0          nompi_py37h513d04c_100    conda-forge
hdf5                      1.10.5          nompi_h3c11f04_1103    conda-forge
icu                       58.2                 h211956c_0    anaconda
idna                      2.8                      pypi_0    pypi
igraph                    0.7.1                         4    conda-forge
intel-openmp              2019.4                      243    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
joblib                    0.13.2                     py_0    conda-forge
jpeg                      9b                   h024ee3a_2    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
jsonpatch                 1.24                     pypi_0    pypi
jsonpointer               2.0                      pypi_0    pypi
libblas                   3.8.0               12_openblas    conda-forge
libboost                  1.67.0               h46d08c1_4    anaconda
libcblas                  3.8.0               12_openblas    conda-forge
libedit                   3.1.20181209         hc058e9b_0    defaults
libffi                    3.2.1                hd88cf55_4    defaults
libgcc-ng                 9.1.0                hdf63c60_0    defaults
libgfortran-ng            7.3.0                hdf63c60_0    defaults
libiconv                  1.15              h516909a_1005    conda-forge
liblapack                 3.8.0               12_openblas    conda-forge
libopenblas               0.3.7                h6e990d7_1    conda-forge
libpng                    1.6.37               hed695b0_0    conda-forge
libstdcxx-ng              9.1.0                hdf63c60_0    defaults
libtiff                   4.0.10               h2733197_2    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
libuuid                   1.0.3                h1bed415_2    anaconda
libxcb                    1.13              h14c3975_1002    conda-forge
libxml2                   2.9.9                hea5a465_1    anaconda
mkl                       2019.4                      243    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
ncurses                   6.1                  he6710b0_1    defaults
ninja                     1.9.0            py37hfd86e86_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
numpy                     1.17.2                   pypi_0    pypi
olefile                   0.46                     py37_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
openssl                   1.1.1                h7b6447c_0    anaconda
pcre                      8.43                 he6710b0_0    anaconda
pillow                    6.1.0                    pypi_0    pypi
pip                       19.2.3                   py37_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
pixman                    0.38.0            h516909a_1003    conda-forge
pthread-stubs             0.4               h14c3975_1001    conda-forge
py-boost                  1.67.0           py37h04863e7_4    anaconda
pycairo                   1.18.1           py37h2a1e443_0    anaconda
pycparser                 2.19                     py37_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
pynvrtc                   9.2                      py37_0    qwant
python                    3.7.4                h265db76_1    defaults
python-igraph             0.7.1.post7      py37h516909a_0    conda-forge
pytorch                   1.2.0           py3.7_cuda10.0.130_cudnn7.6.2_0    pytorch
pyzmq                     18.1.0                   pypi_0    pypi
readline                  7.0                  h7b6447c_5    defaults
requests                  2.22.0                   pypi_0    pypi
[... 4 lines cut ...]
sqlite                    3.29.0               h7b6447c_0    defaults
tk                        8.6.8                hbc83047_0    defaults
torch                     1.2.0                    pypi_0    pypi
[... cut ...]
xorg-libxrender           0.9.10            h516909a_1002    conda-forge
xorg-renderproto          0.11.1            h14c3975_1002    conda-forge
xorg-xextproto            7.3.0             h14c3975_1002    conda-forge
xorg-xproto               7.0.31            h14c3975_1007    conda-forge
xz                        5.2.4                h14c3975_4    defaults
zlib                      1.2.11               h7b6447c_3    defaults
zstd                      1.3.7                h0b5b093_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
```

CharlieLeee, 2019-10-15:
@zeroAska Hi, I’m currently struggling with this 
```
ImportError: ./partition/cut-pursuit/src/libcp.so: undefined symbol: _ZN5boost6python6detail11init_moduleER11PyModuleDefPFvvE
```
issue, and I would be grateful if you could tell me specifically the way you solved the issue, many thanks!

loicland, 2019-10-15:
Did you try to set the li k tk boost by hand as detailed [here](https://github.com/BVLC/caffe/issues/5810#issue-246590926)?
My guess would be that the linker finds boost for the wrong version of python.

CharlieLeee, 2019-10-15:
@loicland Thanks for the follow up! 
This is the CMakeList file:

```

##############################
### Find required packages ###
##############################
find_package(PythonLibs)
find_package(PythonInterp)
#include(FindNumPy)
include_directories(../include)

#find_package(Boost 1.65.0 COMPONENTS graph REQUIRED PATHS /home/undergraduate/miniconda2/envs/spg/lib) #system filesystem thread serialization
#if (${Boost_MINOR_VERSION} LESS 67 )
# find_package(Boost 1.65.0 COMPONENTS numpy${PYTHON_VERSION_MAJOR} REQUIRED PATHS /home/undergraduate/miniconda2/envs/spg/lib) #system filesystem thread serialization
#else()
# set(PYTHONVERSION ${PYTHON_VERSION_MAJOR}${PYTHON_VERSION_MINOR})
# find_package(Boost 1.67.0 COMPONENTS numpy${PYTHONVERSION} REQUIRED PATHS /home/undergraduate/miniconda2/envs/spg/lib)
#endif()

set(Boost_INCLUDE_DIRS /home/undergraduate/miniconda3/envs/sp/include/boost)
set(Boost_LIBRARY_DIRS /home/undergraduate/miniconda3/envs/sp/lib)

include_directories(${Boost_INCLUDE_DIRS})
link_directories(${Boost_LIBRARY_DIRS})

message("PYTHON LIBRARIES ARE " ${PYTHON_LIBRARIES})
INCLUDE_DIRECTORIES(${PYTHON_INCLUDE_DIRS} ${PYTHON_NUMPY_INCLUDE_DIR})
LINK_DIRECTORIES(${PYTHON_LIBRARY_DIRS})

file(GLOB CP_HEADERS include/*.h)

set(CMAKE_LD_FLAG "${CMAKE_LD_FLAGS} -shared -Wl -fPIC --export-dynamic -o -O3 -Wall -std=c++11")
# message(${Boost_LIBRARIES})
add_library(cp SHARED cutpursuit.cpp ${CP_HEADERS})
target_link_libraries(cp ${Boost_LIBRARIES} ${PYTHON_LIBRARIES}) 
```

Since I have already comment out the “finding boost package process” and instead setting the directory manually, so I’m not sure creating a soft link will actually work. However, since I’ve also commented out to avoid errors in the CMake process
```
message(${Boost_LIBRARIES})
```
so I would like to know if “Boost_LIBRARIES” is the same as “Boost_LIBRARY_DIRS”? Or is a directory found by the findpackage function?

CharlieLeee, 2019-10-16:
@loicland would you please tell me the difference between ${Boost_LIBRARIES} and Boost_LIBRARY_DIRS?

zzusunquan, 2019-10-31:
> 通过将Boost_LIBRARIES设置为libboost_numpy3。*。so路径解决了问题

I meet the same error as you , could you gei me a detail steps to solve the problem ,
方便加个联系方式沟通一下吗 我的微信sq8262109