loicland/superpoint_graph #62: To visualize intermediary steps

leemengwei, 2018-11-09:
Hi, your works are great.
Here's a question: by running visualize.py, I want to have a visualized image of intermediary steps, yet I only find generated files like (in clouds/ directory):
conferenceRoom_1_err.ply   conferenceRoom_1_GT.ply         conferenceRoom_1_pred.ply     conferenceRoom_1_rgb.ply
How can I visualize them?

leemengwei, 2018-11-15:
With the help of Jaily, I know that the generated ply file can be visualized by software named 'meshlab', Which is great, better mention it, thank you.

closed 2018-11-15 by leemengwei

loicland, 2018-11-15:
Hi,

Sorry i forgot to answer this one. I recommend Cloud Compare for visualizing point clouds as well, which is more powerful and has more option than meshlab. But you can already do a lot with meshlab.