loicland/superpoint_graph #63: Perhaps a mistake in class index?

leemengwei, 2018-11-09:
Is it a mistake in class index?
The Semantic8 dataset, is consist of {1: man-made terrain, 2: natural terrain, 3: high vegetation, 4: low vegetation, 5: buildings, 6: hard scape, 7: scanning artefacts, 8: cars} and an additional label {0: unlabeled points} as documented in it's website. While in learning/sema3d_dataset.py, line 66, I find that the class index turn out to be: {0:'terrain_man', 1:'terrain_nature', 2:'veget_hi', 3:'veget_low', 4:'building', 5:'scape', 6:'artefact', 7:'cars'}, all one index shift forward. Is't supposed to be like that? thankyou!

leemengwei, 2018-11-15:
The later refers to the class index, so it's one less. 
Problems solved with sparse automatic lidar data. So I'm closing this issue. Further details if anyone may be interested with:
My data set is of:
classid = {"DontCare": 0, "cyclist": 1, "tricycle": 2, "smallMot":3, \
           "bigMot": 4, "pedestrian": 5, "crowds": 6, "unknown": 7}
Yet 'DontCare' class takes about 95% percent of whole data set. At beginning, I wrongly take the 'DontCare' class as background class in this SPG project, then almost all points are predicted to be 'smallMot' (the second dominant class) and the loss drops to zero, giving me a meaningless model (because the SPGnet don't treat background class with loss).
Then I just make my 'DontCare' class to additional class, say class 8, such act would force the network to predict 'DontCare' class to be an actual meaningful class with loss counting on, and it works fine.

closed 2018-11-15 by leemengwei

luoxiaoliaolan, 2018-11-29:
@leemengwei 
Have you solved the mistake in class index? I also encountered the mistake. This led to the results that the 0th and 1st categories were inseparable. Just like this:
[image]
The color I set for the classes is ("ground": 0, "building": 1, "tree": 2, "pole":3). But I got the result showed that the mistake colors. 
[image]
Do you know where should I change to solve this problem?
Welcome to send an email to my email address. (lybluoxiao@qq.com)

loicland, 2018-11-29:
@ leemengwei sorry I seem to have missed your issue earlier.

There is indeed a discrepancy in how the classes are indexed in the training set and how they are predicted, I can see how that would be confusing.

When reading the data, the class `0` is reserved for unlabelled data. For inference, all superpoints will be associated with a semantic class, such that none will have the unlabelled class. So we have shifted the classes by one, so that the first real class (`1` in the data) will be associated with `0`.

This `0` class is *not* designed for a background class, but rather for small regions of annotated data, or when the annotation is on a subsampled version of the full graph. If you put all your background in the class `0` then at inference time it will see all those superpoints of unknown shape (unlabelled data are excluded from the supervision) and will try to classify them as the closest real class. Using an actual class for background is indeed the way to go.

loicland, 2018-11-29:
@luoxiaoliaolan 

hi,

your problem is that you put the class ground as `0` (i.e. unannotated), so it is ignore when supervising. Furthermore, it will shift all the classes by one, meaning that the classes are uncorrectly labeleld as well.

What you should do is that the fucntion yiou use to read your data should associate the following labels:

"ground": 1, "building": 2, "tree": 3, "pole":4

and it should work! Let me know if you still encounter problems.

luoxiaoliaolan, 2018-11-29:
@loicland 
Thank you very much!!
I would like to know if the modification you mentioned is to modify the category label in the original file(XXX.labels) or when the program reads the data.
My dataset like this:
[image]
[image]
[image]

labels file:
[image]

loicland, 2018-11-29:
You can either change your .labels files so that your classes starts at 1 (provided you dont have unannotated data), or change it in the read_my_dataset function in provider.py if you coded your own.