loicland/superpoint_graph #66: S3DIS model Inference on other dataset

nicolas-chaulet, 2018-11-19:
Hi Loic,
Thanks a lot for providing such a great project. I would like to use an SPG model trained on the S3DIS dataset to segment some indoor scenes of my own. The point clouds we collect do not have colour information and I trained a new SPG model on the S3DIS dataset without colour information with no problem on the cross validation split number 1. I get some good results on train and test sets which is good. 

Things get a little more troublesome when I use this model to segment my own point clouds, it performs not well at all by mi classifying a lot more points than expected inside the room, see some results below:
[image]

It misses the table, chairs and sofa when they definitely look similar to the ones in the S3DIS dataset. I know it's a big ask but I was wondering if you had an idea of what could have gone wrong there?
The model, training logs and various visualisation output can be found there:
https://drive.google.com/drive/folders/1M_e3XW_72NfqxEbnQFCo7Ubd2Ch8XoMG?usp=sharing
 
Thanks in advance,
Nicolas

nicolas-chaulet, 2018-11-20:
Hi Loic,
Thanks a lot for you answer, there is no rush and I really appreciate your interest and support. Yes, I am doing the two things you suggest, I just double checked and it looks just fine to me. One thing is that the surfaces in our dataset are a lot less clean than in the S3DIS dataset. Walls have thickness, same for table top for example. I will try to smooth it out as a preprocessing step and see if that helps.

loicland, 2018-11-20:
In which case I would:
- retrain without the lpsv values which will be greatly affected by the geometric noise
- increase the jitter on S3DIS to make it look more like yours (line 244 of `learning/spg.py`). Careful with this parameter, if too high the pointNets won't learn anything useful.

nicolas-chaulet, 2018-11-28:
Hi Loic,
Quick update on this case. I am quite certain now that the issue is related to the way I prepare the data and specifically to the scaling that is applied to the edge features (line 37 in `learning/s3dis_dataset.py`). I did not pay attention to this too much but now it is obvious that we need to scale the edge features using the same scaling factors as the one used during training. I still need to do some more testing but I really think this is the source of my troubles.

loicland, 2018-11-29:
Glad to hear it.

Note as well that the rooms in S3DIS are aligned with either the x or y axis. Try to do the same for your dataset, especially if you use XYZ as point features.

nicolas-chaulet, 2018-11-29:
I was actually just writing to you. It was definitely the sedge feature normalisation that created the issue. It even does ok on noisy data.
[image]

Yes, I saw that the dataset is aligned. I actually would like to get rid of this concept of rooms, in my experience it's actually complicated to get this kind of information in the first place from field data.

closed 2018-11-29 by nicolas-chaulet