loicland/superpoint_graph #76: Run inference without training set

lance6716, 2018-12-16:
I'm testing an idea, and only want to run your pertained models on a rbg-ply. but after some modifications on code I guess it's tightly coupled on both training set and testing set. what's the most convenient way to run inference?

lance6716, 2018-12-17:
Need to run learning/custom_dataset.py to get files in `parsed` folder, maybe that's a step missed from doc.

I'm trying to download only one sema3d point-cloud file and it's label, putting them into my `train` folder and modifying code to use proper functions to get xyz and rgb from file of different format. Hope this would work

loicland, 2018-12-17:
Hi,

yes the data needs to be parsed before inference or learning. Easiest way to run inference is to copy the trained models in your /results/ folder and then run `main.py` with the `--resume RESUME` argument and `--epochs -1`.

lance6716, 2018-12-18:
> Hi,
> 
> yes the data needs to be parsed before inference or learning. Easiest way to run inference is to copy the trained models in your /results/ folder and then run `main.py` with the `--resume RESUME` argument and `--epochs -1`.

I did these to make it run. 

1. Download bildstein_station1_xyz_intensity_rgb.7z and its label from sema3d, put them under `datasets/custom_set/data/train/`. Put my ply under `datasets/custom_set/data/test`

2. in `partition/partition.py`, change `n_labels` of `custom_dataset` to 8, and add logic to deal with different point cloud type. For training data, `read_semantic3d_format` is used to read xyz, rgb, label. For testing data, `read_ply` is used to read xyz and rgb, then labels is set to `[]` after `libply_c.prune`

3. `python partition/partition.py --dataset custom_datase
t --ROOT_PATH /root/projects/superpoint_graph/datasets/custom_set --voxel_width
0.05 --reg_strength 0.8 --ver_batch 5000000`

4. `python learning/custom_dataset.py`

5. in `learning/main.py`, comment out computaions related to confusion_matrix, `eval_final` only returns predictions, and write it to h5 file.

6. 
```shell
CUDA_VISIBLE_DEVICES=0,1 python learning/main.py --dataset custom_dataset --db_test_name test --epochs -1 --lr_steps '[350, 400, 450]' --test_nth_epoch 100 --model_config 'gru_10,f_8' --ptn_nfeat_stn 11 --nworkers 2 --odir "results/sema3d/trainval_best" --resume RESUME

python partition/visualize.py --dataset sema3d --ROOT_PATH datasets/custom_set/ --res_file 'results/sema3d/trainval_best/predictions_test' --file_path 'test/output' --output_type ifprs
```
And there's other modifications on changing file names. I'm not sure about whether it's running as expected, did I missed something? The partition point-cloud looks well but in prediction point-cloud, most buildings are labeled as 'hard scape'.

loicland, 2018-12-18:
Hi,

it seems good. The errors may come from different normalization in your data compared to S3DIS, as mentioned by @nicolas-chaulet. Make sure that:
- your xyz values have the same units than Semantic3D (meters I think)
- your z value is comparable (road around -1, buildings around 4-5)
- your RGB values have the same unit (uint8)
- did you add the scaler01 function for your custom set?

Can you post an image of (a part of) your target cloud?

lance6716, 2018-12-21:
> Hi,
>
> it seems good. The errors may come from different normalization in your data compared to S3DIS, as mentioned by @nicolas-chaulet. Make sure that:
>
> * your xyz values have the same units than [...]

Sorry for the late reply. I checked my pointcloud, it doesn't have a comparable scale, I'll run another inference some days later. Merry Christmas!

[image]

loicland, 2018-12-21:
Hi,

the scaling is the problem. Maybe try to scale/shift your data such that it overlaps with Semantic3D data when you load both point cloud into your viewer (length in meters, ground level at -1). Of note, thoshigh rise buildings are quite different from the urban scenes of Semantic3D, I have no ieda how the tained modelwill interpret them.

Furthermore, Semantic3D is acquired with a dense scan which does not seem to be the case of your data. Again, there might be a generalization problem here.

Anyway, let me know how this goes, and merry xmas to you!

lance6716, 2018-12-26:
Below are pictures of results cloud in cloud folder, output_rgb, output_partition, output_pred. Have checked ground level at -1 meter and other scale problem. I think some buildings in center of pictures can't be recognized well somehow, and the roof of building sometimes separate from the main part.

[image]
[image]
[image]

loicland, 2018-12-26:
Hi,

It is normal for the roofs to be in separate components, as their verticality vakue differs from the facades. Your dataset seems too different from semantic 3d (both structurally and acquisition-wise). You should at least fine tune the models on annotated data.

Furthermore you ahould decrease the regularization strength as the components are probably too big.

lance6716, 2018-12-26:
ah, could you recommend some annotation tools for point-cloud? I guess
that's a hard work.

loicland, 2019-01-10:
Sorry no, I am not aware of free annotation tools.