loicland/superpoint_graph #77: Any possibility to add tensorboardX graph module in the code?

staochun, 2018-12-28:
Thanks to your excellent SPG, I have got good predictions and results for my research. And I really want to visualize the resulting graph generated by SPG.
TensorBoardX is a useful package to achieve that task, which can be compatible with Torch.  And also I have successfully visualized the learning rate and other scalar variables. However, when trying to visualize the graph, I confront some difficulties about what is the input of the function add_graph in TensorBoardX.FileWriter class. It's required that the input must be tuples, lists and cannot be the string (from The feedback of Traceback), and must be converted into Variable type to fit the input of function (from The official guide of TensorBoardX). 
And because you use torch.dataloader to process the preliminary cloud data, we don't know what is the real variable as input to the model and almost try all variables I think related to the right input, including cloud_data and so on. And we even check the type of variables and try to compulsorily convert into what fits the input, for example, (cloud_data, ). 
However, all fail. So I would like to ask if you are willing to make some improvement for the visualization of SPG model, (even if you don't want to use TensorBoardX and implement in other ways), or at least, could you please help me some with this challenging problem? Thanks a lot for your sincere help!

loicland, 2018-12-28:
Hi,

First glad to hear SPG works well for your project. However I am not familiar at all with tensorboardx. From what I gathered, graph vizualization will represent the entire neural network in readable graph form. For SPG, this graph can be quite complicated, and will be different for each input point cloud, and even for each training iteration. Indeed, the superpoints connectivity will depend on the subsampled adjacency graph between components. Furthermore the resulting graph can be quite large, especially since the message passing operates on the graph duplicated ten times.

I am unsure what insight you expect of this. Maybe vizualizing the graph for the pointnet-like superpoint embedding network and the edge features embedding would be enough for your requirements?

Anyway I will be happy to answer any clarification that you might need for variables and structure.

Loic

staochun, 2018-12-30:
Wow. Really appreciate your kind help and quick reply. Maybe I just want to visualize the edge features embedding to see what is going on my custom dataset. Do you know some methods to achieve that? And also it can be anticipated that the dataset will generate a graph with lots of superpoints and superedges, if I want to make a hierarchical analysis to see which edges play a more (or the most )important role in the graph (I have already visualized the graph and points with "ply" format using your code, and thanks a lot), do you have some genius ideas to achieve that?

closed 2019-04-23 by loicland