loicland/superpoint_graph #79: error if I use "--db_test_name val --db_train_name train" as the training parameter

cbyzju, 2019-01-09:
Hi， I really appreciated your great work. When I adapt your model to train my custom dataset, I use parameter " --db_test_name test --db_train_name trainval", everything is fine, I got the correct result, while I think if I use the above parameters, there is no validation step during model train. So I change the training parameter to " --db_test_name val --db_train_name train", I got the following errors when make the first evaluation, could you give me some kind help?

`File "learning/main.py", line 387, in <module>
    main()
  File "learning/main.py", line 294, in main
    acc_test, oacc_test, avg_iou_test, avg_acc_test = eval()
  File "learning/main.py", line 223, in eval
    outputs = model.ecc(embeddings)
  File "/home/cby/anaconda3/envs/python3.5/lib/python3.5/site-packages/torch/nn/modules/module.py", line 224, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/cby/Desktop/HDMap/learning/graphnet.py", line 97, in forward
    input = module(input)
  File "/home/cby/anaconda3/envs/python3.5/lib/python3.5/site-packages/torch/nn/modules/module.py", line 224, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/cby/Desktop/HDMap/learning/modules.py", line 42, in forward
    weights = self._fnet(edgefeats)
  File "/home/cby/anaconda3/envs/python3.5/lib/python3.5/site-packages/torch/nn/modules/module.py", line 224, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/cby/anaconda3/envs/python3.5/lib/python3.5/site-packages/torch/nn/modules/container.py", line 67, in forward
    input = module(input)
  File "/home/cby/anaconda3/envs/python3.5/lib/python3.5/site-packages/torch/nn/modules/module.py", line 224, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/cby/anaconda3/envs/python3.5/lib/python3.5/site-packages/torch/nn/modules/batchnorm.py", line 37, in forward
    self.training, self.momentum, self.eps)
  File "/home/cby/anaconda3/envs/python3.5/lib/python3.5/site-packages/torch/nn/functional.py", line 639, in batch_norm
    return f(input, weight, bias)
RuntimeError: CUDNN_STATUS_NOT_SUPPORTED. This error may appear if you passed in a non-contiguous input.`

loicland, 2019-01-10:
Hi,

How far as I can tell this error happens when the input of some layers gets too big, and is a cudnn bug (see this [issue](https://github.com/pytorch/pytorch/issues/4107)).

Are you using this on a custom datasets? If so it could be that one the file in the validation set is too large. Maybe try to increase upsampling?

loicland, 2019-01-10:
Hi,

How far as I can tell this error happens when the input of some layers gets too big, and is a cudnn bug (see this [issue](https://github.com/pytorch/pytorch/issues/4107)).

Are you using this on a custom datasets? If so it could be that one of the superpoint graphs in the validation set is too large (more than 64k superpoints I think).

To fix the issue you could try re-running the partition step with a higher reg_strength to get coarser graphs. If it is not an option for you, there is another way around it. Let me know if you need this fix urgently, or I will just try to add it with the next patch.

cbyzju, 2019-01-10:
yeah! That's the reason. the size of all my train superpoint graphs range within 50~100 MB, it's very large. Considering your great idea, I will split the validation file to avoid cudnn bug, and I think this issue can be closed now, thank you very much.

closed 2019-01-10 by cbyzju