loicland/superpoint_graph #80: RuntimeError: Dimension out of range (expected to be in range of [-1, 0], but got 1)

andreac, 2019-01-11:
Hi,
I'm testing with s3dis dataset. When i run 

`for FOLD in 0 1 2 3 4; do \
CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset s3dis --S3DIS_PATH ./dataset --cvfold $FOLD --epochs 350 --lr_steps '[275,320]' \
--test_nth_epoch 50 --model_config 'gru_10_0,f_13' --ptn_nfeat_stn 14 --nworkers 2 --odir "results/dataset/best/cv${FOLD}"; \
done`

it start running and after a bit i get this error:
`Epoch 0/350 (results/dataset/best/cv4):
  0%|                                                                        | 0/20 [00:00<?, ?it/s]/home/spxstrix/DATI/ENEL/retineurali/superpoint_graph/learning/pointnet.py:131: UserWarning: volatile was removed and now has no effect. Use `with torch.no_grad():` instead.
  out = model.ptn(Variable(clouds, volatile=True), Variable(clouds_global, volatile=True))
/home/spxstrix/miniconda3/lib/python3.7/site-packages/torch/nn/functional.py:1006: UserWarning: nn.functional.sigmoid is deprecated. Use torch.sigmoid instead.
  warnings.warn("nn.functional.sigmoid is deprecated. Use torch.sigmoid instead.")
Warning: source/destination slices have same size but different shape for an index operation.  This behavior is deprecated.
learning/main.py:196: UserWarning: invalid index of a 0-dim tensor. This will be an error in PyTorch 0.5. Use tensor.item() to convert a 0-dim tensor to a Python number
  loss_meter.add(loss.data[0])
Exception ignored in: <function _DataLoaderIter.__del__ at 0x7f8896cfa620>
Traceback (most recent call last):
  File "/home/spxstrix/miniconda3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 399, in __del__
    self._shutdown_workers()
  File "/home/spxstrix/miniconda3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 378, in _shutdown_workers
    self.worker_result_queue.get()
  File "/home/spxstrix/miniconda3/lib/python3.7/multiprocessing/queues.py", line 354, in get
    return _ForkingPickler.loads(res)
  File "/home/spxstrix/miniconda3/lib/python3.7/site-packages/torch/multiprocessing/reductions.py", line 151, in rebuild_storage_fd
    fd = df.detach()
  File "/home/spxstrix/miniconda3/lib/python3.7/multiprocessing/resource_sharer.py", line 57, in detach
    with _resource_sharer.get_connection(self._id) as conn:
  File "/home/spxstrix/miniconda3/lib/python3.7/multiprocessing/resource_sharer.py", line 87, in get_connection
    c = Client(address, authkey=process.current_process().authkey)
  File "/home/spxstrix/miniconda3/lib/python3.7/multiprocessing/connection.py", line 498, in Client
    answer_challenge(c, authkey)
  File "/home/spxstrix/miniconda3/lib/python3.7/multiprocessing/connection.py", line 746, in answer_challenge
    response = connection.recv_bytes(256)        # reject large message
  File "/home/spxstrix/miniconda3/lib/python3.7/multiprocessing/connection.py", line 216, in recv_bytes
    buf = self._recv_bytes(maxlength)
  File "/home/spxstrix/miniconda3/lib/python3.7/multiprocessing/connection.py", line 407, in _recv_bytes
    buf = self._recv(4)
  File "/home/spxstrix/miniconda3/lib/python3.7/multiprocessing/connection.py", line 379, in _recv
    chunk = read(handle, remaining)
ConnectionResetError: [Errno 104] Connection reset by peer
Traceback (most recent call last):
  File "learning/main.py", line 387, in <module>
    main()
  File "learning/main.py", line 286, in main
    acc, loss, oacc, avg_iou = train()
  File "learning/main.py", line 199, in train
    acc_meter.add(o_cpu, t_cpu)
  File "/home/spxstrix/.local/lib/python3.7/site-packages/torchnet/meter/classerrormeter.py", line 38, in add
    pred = torch.from_numpy(output).topk(maxk, 1, True, True)[1].numpy()
RuntimeError: Dimension out of range (expected to be in range of [-1, 0], but got 1)
`

Previusly steps ends fine, could you help me?

andreac, 2019-01-14:
I think the problem is about the configuration of pythorch e torchnet. I'm using Ubuntu 18.04
Cuda 9.2.
I'm tring different setup. Now i've intalled  pytorch 0.3 and torchnet with pip install torchnet.

I have the following error:
`
Traceback (most recent call last):
  File "learning/main.py", line 387, in <module>
    main()
  File "learning/main.py", line 286, in main
    acc, loss, oacc, avg_iou = train()
  File "learning/main.py", line 199, in train
    acc_meter.add(o_cpu, t_cpu)
  File "/home/spxstrix/DATI/ENEL/retineurali/superpoint/lib/python3.6/site-packages/torchnet/meter/classerrormeter.py", line 38, in add
    pred = torch.from_numpy(output).topk(maxk, 1, True, True)[1].numpy()
RuntimeError: the given numpy array has zero-sized dimensions. Zero-sized dimensions are not supported in PyTorch
`

andreac, 2019-01-15:
Hi, the problem is about the dataset i'm using. i have  adataset based on txt but whitout rgb value.
The problems depends on it

closed 2019-01-15 by andreac

yaping222, 2019-01-30:
Hi andreac, I met the same problem. Could you give some suggestions on how to solve it?

loicland, 2019-01-30:
Hi. Can you give more context? Is the error happening line 199 of main.py as well? Does your dataset lacks rgb values as well?

yaping222, 2019-02-01:
> Hi. Can you give more context? Is the error happening line 199 of main.py as well? Does your dataset lacks rgb values as well?

Yes, the error was in line199. I don't have rgb values in my dataset so I just set them as zeros and got the error: 
RuntimeError: the given numpy array has zero-sized dimensions. Zero-sized dimensions are not supported in PyTorch

Then I remove the rgb values, the code works.

Thank you very much for your reply!