loicland/superpoint_graph #82: Add Example from dataset without RGB

andreac, 2019-01-15:
Hi,
I'm working on a dataset without RGB, and i'm not able to train. I follow your suggestion, it have a similar configuration son s3dis dataset (txt file divided by class).

in partition.py i modify like that:
`xyz, rgb, labels = read_s3dis_format(data_file)
                if args.voxel_width > 0:
                    xyz = libply_c.prune(xyz, args.voxel_width, np.zeros(xyz.shape,dtype='u1'), np.zeros(xyz.shape,dtype='u1'), 0)[0]
                rgb=[]
                labels=[]`

i modified n_labels and all other stuff but now i0m getting an error on 

` self.confusion_matrix[:,predicted[i]] += ground_truth_vec[i,:]
IndexError: index 9 is out of bounds for axis 1 with size 7`

Could you help adding an example wihtout RGB? i can provide the dataset

loicland, 2019-01-15:
If your data has 7 labels, n_labels should be set to 7. But it seems like `predicted` goes up to 9?

Can you check the range of `predicted` and `ground_truth_vec`? Also can you check in the superpointgraphs h5 files the shape of the `sp_labels` field ? It should be 9, the first column for the spsize, the second for the unlabelled data and the rest for the labels breakdown.

andreac, 2019-01-15:
Yes you are right, this is my data:
`number_of_labels: 8`
```
ground_truth_vec[0,:]
[0 0 0 0 0 0 0 0]
predicted[0]
5
self.confusion_matrix[:,predicted[0]]
[0. 0. 0. 0. 0. 0. 0. 0.]

```
Traceback (most recent call last):
  File "learning/main.py", line 398, in <module>
    main()
  File "learning/main.py", line 295, in main
    acc, loss, oacc, avg_iou = train()
  File "learning/main.py", line 209, in train
    confusion_matrix.count_predicted_batch(tvec_cpu, np.argmax(o_cpu,1))
  File "/home/spxstrix/DATI/ENEL/retineurali/superpoint/superpoint_graph/learning/metrics.py", line 28, in count_predicted_batch
    self.confusion_matrix[:,predicted[i]] += ground_truth_vec[i,:]
IndexError: index 10 is out of bounds for axis 1 with size 8
```
I think i'm missing some values or other stuff but i'm not able to find where
thanks

andreac, 2019-01-16:
i fixed the shape of sp_labels. Now it print that:
```
spezzata_0.h5
(8248, 9)
```

but when i run main.py i'm getting the same error:

```
o_cpu
[12 10  0 ...  1  1  1]
ground_truth_vec[0,:]
[13  0  0  0  0  0  0  0]
predicted[0]
12
self.confusion_matrix[:,predicted[0]]
Traceback (most recent call last):
  File "learning/main.py", line 398, in <module>
    main()
  File "learning/main.py", line 295, in main
    acc, loss, oacc, avg_iou = train()
  File "learning/main.py", line 209, in train
    confusion_matrix.count_predicted_batch(tvec_cpu, np.argmax(o_cpu,1))
  File "/home/spxstrix/DATI/ENEL/retineurali/superpoint/superpoint_graph/learning/metrics.py", line 24, in count_predicted_batch
    print(self.confusion_matrix[:,predicted[0]])
IndexError: index 12 is out of bounds for axis 1 with size 8

```

any idea?

loicland, 2019-01-19:
Sorry for delay.

Can you print the shape of `o_cpu` and `ground_truth_vec` as well as a example of `sp_labels` in a `supepoint_graph.h5` file?

In short I think there is an issue with the labels in your dataloader. If you have 7 classes `o_cpu` and `ground_truth_vec` should not go up to 12/13.

andreac, 2019-01-20:
I deleted the project and restarted with a new one and tadaaaa now it works! Fantastic! I have a meeting Monday so I can show results!
I have other question, is possible to merge al results model to create only one model? 
Thanks a lot!

loicland, 2019-01-20:
Do you mean one file (.ply, .las or other) with all your data? If so it should be pretty straightforward with pytfile . See my ply-manipulating functions in `partition/provider.py`.

andreac, 2019-01-21:
Now i have one model for every folder, I mean create an only .h5 file with the models of all folder trained. 
For example merge ol .h5 from superpointgraph and features to create only one model

[image]

loicland, 2019-01-21:
I am still unsure of what you mean. By model do you mean feature file, and not trained neural weight? If so you can look into `/learning/spg.eccpc_collate`, but I am unsure why you would want to do that. Very large models are bound to require too much memory, even for inference.

andreac, 2019-01-21:
I mean trained neural weight to build only one model to classified all my pointclouds.
I have 500Gb of ram and 8 GPU's, maybe it's enough

loicland, 2019-01-22:
Models are saved in `.tar` and not `.h5`, so I am still unsure what you mean.

If you want to have one trained model (neural weights) trained on all your data it should be a rather straightforward modification of the `get_datasets` function in the `py` file preparing your custom dataset  in `/training`.

andreac, 2019-01-22:
I trained in all my data, the result directory contains:

- model.pth.tar
- predictions_val.h5

to test i use visualize.py that in this line search for the prediction of my file

```
pred_red  = np.array(h5py.File(res_file, 'r').get(folder + file_name))
```
is it possible to predict on a pointcloud that i don't use to test or train?

closed 2019-04-23 by loicland