loicland/superpoint_graph #83: Training on Custom dataset

TotoLulu94, 2019-01-21:
Hello,
I am trying to retrain a model from scratch without using the RGB values using a custom dataset really close to S3DIS (instead of entire room, i'm using partial room by cropping it vertically). I am having incoherent result between the training accuracy scores (0.9 for training accuracy and 0.8 for testing for testing accuracy) and the results I have when trying to run inferences on the same testing dataset used during the training process : By trying a few random point clouds from the testing set, all the floors are recognized as ceiling, the wall as floors...
My dataset has 14 different classes and annotates the first class (let's say ceiling) by 0. I read in another issues that the label 0 is reserved for unannotated data (that I don't have in my dataset). But since I modified my label reader accordingly, using 0 for ceiling should be fine right ?
Here's the command line I used for training : 
` python learning/main.py \
        --dataset 'custom_s3dis' \
        --S3DIS_PATH 'data/custom_S3DIS' \
        --cvfold 1 \
        --epochs 350 \
        --lr_steps '[275,320]' \
        --test_nth_epoch 50 \
        --model_config 'gru_10_0,f_14' \
        --ptn_nfeat_stn 8 \
        --pc_attribs 'xyzelpsv' \
        --nworkers 8 \
        --odir 'results/s3dis/bw/cv1_4' `
I believe that my script partition/partition.py creates the features files and spg files correctly.
Thank you for your work on SPG !

loicland, 2019-01-22:
Hi,
a few answers:
- if your dataset is composed of partial rooms, you should retrain without using the scaled position features (option `--pc_attribs xyzrgbelpsv`)
- it is expected to have difference between training and testing set value. If it is an issue for you, try to add drop out layers (option `--ptn_prelast_do 0.5` for example)
- if you mean that the inference behave differently from train to eval, this is surely due to the batchnorms. Either use a larger dataset for more epochs, or replace the batchnorms by group norms
- 0 is reserved for unannotated data. Even if you don't have them, add 1 to the labels in your data reading function. the points will be labeled starting from 0 for your first class.

closed 2019-01-22 by TotoLulu94

ghgr, 2019-02-25:
Hi,
 
I'm confused by the fact that 0 is reserved for unannotated data. If that's the case, I don't understand why the files *learning/sema3d_dataset.py* and *learning/s3dis_dataset.py* show respectively:

```python
'inv_class_map': {0:'terrain_man', 1:'terrain_nature', 2:'veget_hi', 3:'veget_low', 4:'building', 5:'scape', 6:'artefact', 7:'cars'},
```
and
```python
'inv_class_map': {0:'ceiling', 1:'floor', 2:'wall', 3:'column', 4:'beam', 5:'window', 6:'door', 7:'table', 8:'chair', 9:'bookcase', 10:'sofa', 11:'board', 12:'clutter'},
```

Are in these datasets the class 0 *undefined* or not? I'm trying to load a custom pointcloud and I reserved the class 0 for *undefined*, but then I'm not sure how to numerate the *inv_class_map* field.

loicland, 2019-02-25:
Right, I admit it is a little confusing.

The class "0" is reserved for unannotated  **in the ground truth files**. Since the network will associate class labels to all points (and hence will never output 0/unannotated labels), and since the unannotated points do not appear in the evaluation metrics, we have shifted the class indexes by one (the output class indexes starts at zero). Clear?

ghgr, 2019-02-25:
Thanks for your answer!

So, to make sure I understood, if my custom, ground truth pointcloud has following labels:

1: sofa
2: lamp
3: roof
....
(nothing for 0, since no point is **unknown**)

Then, in the **inv_class_map** field I should write them like this?
```python
'inv_class_map': {0:'sofa 1:'lamp', 2:'roof', ...}
```

loicland, 2019-02-25:
yep!

ghgr, 2019-02-25:
Thanks a lot!