loicland/superpoint_graph #87: Problem related CUDA driver when training

hagianga21, 2019-02-13:
Hi when I train the network I got this problem. Does any one have suggestion for the the solution? I used Torch 0.3.0, CUDA 9.0 and cupy 5.1.0. Thank you so much

`Epoch 0/500 (results/sema3d/trainval_best):
  0%|                                                                         | 0/7 [00:00<?, ?it/s]Traceback (most recent call last):
  File "learning/main.py", line 387, in <module>
    main()
  File "learning/main.py", line 286, in main
    acc, loss, oacc, avg_iou = train()
  File "learning/main.py", line 184, in train
    outputs = model.ecc(embeddings)
  File "/home/gtruong/anaconda3/envs/super2/lib/python3.6/site-packages/torch/nn/modules/module.py", line 325, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/gtruong/project/superpoint_graph/learning/graphnet.py", line 97, in forward
    input = module(input)
  File "/home/gtruong/anaconda3/envs/super2/lib/python3.6/site-packages/torch/nn/modules/module.py", line 325, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/gtruong/project/superpoint_graph/learning/modules.py", line 54, in forward
    input = ecc.GraphConvFunction(nc, nc, idxn, idxe, degs, degs_gpu, self._edge_mem_limit)(hx, weights)
  File "/home/gtruong/project/superpoint_graph/learning/ecc/GraphConvModule.py", line 67, in forward
    cuda_kernels.conv_aggregate_fw(output.narrow(0,startd,numd), products.view(-1,self._out_channels), self._degs_gpu.narrow(0,startd,numd))
  File "/home/gtruong/project/superpoint_graph/learning/ecc/cuda_kernels.py", line 120, in conv_aggregate_fw
    function, stream = get_kernel_func('conv_aggregate_fw_kernel_v2', conv_aggregate_fw_kernel_v2(), get_dtype(src))
  File "/home/gtruong/project/superpoint_graph/learning/ecc/cuda_kernels.py", line 38, in get_kernel_func
    module.load(bytes(ptx.encode()))
  File "cupy/cuda/function.pyx", line 181, in cupy.cuda.function.Module.load
  File "cupy/cuda/function.pyx", line 183, in cupy.cuda.function.Module.load
  File "cupy/cuda/driver.pyx", line 185, in cupy.cuda.driver.moduleLoadData
  File "cupy/cuda/driver.pyx", line 81, in cupy.cuda.driver.check_status
cupy.cuda.driver.CUDADriverError: CUDA_ERROR_INVALID_PTX: a PTX JIT compilation failed`

closed 2019-02-13 by hagianga21

micaeltchapmi, 2019-04-02:
Hi @hagianga21, 

I'm facing the same issue, were you able to resolve it? If yes, how? I trained normally before and I'm not sure what changes I made that caused this problem. I have been searching and trying different solutions but none of them seems to work. I'm using CUDA 9.0 and pytorch 0.3.1 with cupy 5.3.

mys007, 2019-04-02:
Re-opened in https://github.com/loicland/superpoint_graph/issues/105