loicland/superpoint_graph #88: Disjoint graphs when training on a custom dataset

lferrer, 2019-02-21:
When training on a custom dataset, Line 49 of GraphConvInfo.py will throw an IndexError. I believe that this is due so some graphs being completely disjoint (i.e. having only vertices and no edges). I have been trying to change the regularization term during partition but it hasn't helped. Any suggestions?

loicland, 2019-02-21:
Hi,

See [issue 42](https://github.com/loicland/superpoint_graph/issues/42#issuecomment-460742226) for a discussion on the subject.

In short, it is indeed a documented bug that if the cloud is partitioned into a single superpoint will make the code crash. You could add exceptions at various points to catch this issue, or if you just want a fast prototyping, you can simply detect when it is the case line 214 of `partition.py` and modify by hand the partition to put a single point in its own component.

But first, make sure that this strange partition is indeed intended. Is your point cloud extremely simple?

lferrer, 2019-02-22:
I think that the partition is not working as intended. I have tried with three point clouds having between 2-8 million points each that were produced with photogrammetry software. Would you recommend that I clamp the max length of the super edges or modify any parameters?

loicland, 2019-02-25:
What do you mean not working properly? 8 millions might be a bit high, depending on your hardware. So that I can help you with your problem, could you please:

- upload an exemple of the geof-colored point cloud (option -f in `visualize.py`)
- an exemple of the resulting partition
- the terminal log of the cut pursuit algorithm
- try to increase the pruning (will depend on the sale of your problem, for terrestrial LiDAR 5 cm is usually enough, but for aerial you might want to increase it a lot)

lferrer, 2019-03-04:
Thank you for being willing to help me. I have been trying with different parameters. This is the log of the cut pursuit algorithm with *--voxel_width 50 --reg_strength 0.8* (all other parameters with their default values):

```
L0-CUT PURSUIT WITH L2 FIDELITY
PARAMETERIZATION = FAST
Graph 8924120 vertices and 214178832 edges and observation of dimension 4
Iteration   1 - 10112 components - Saturation   0.0 % - Quadratic Energy 30.100 % - Timer  80.0738
Iteration   2 - 15728 components - Saturation  27.8 % - Quadratic Energy 25.146 % - Timer  214.238
Iteration   3 - 17586 components - Saturation  70.0 % - Quadratic Energy 24.524 % - Timer  325.314
Iteration   4 - 18016 components - Saturation  87.6 % - Quadratic Energy 24.428 % - Timer  396.254
Iteration   5 - 18109 components - Saturation  91.7 % - Quadratic Energy 24.410 % - Timer  450.019
Stopping criterion reached
        computation of the SPG...
Timer : 198.1 / 585.3 / 176.9 
```

I have uploaded the features-colored point cloud and the partition [here](https://drive.google.com/drive/folders/1fuioxU5tk7x6gsN8Z9U5gPkus2jATy8Q?usp=sharing).

loicland, 2019-03-07:
Hi,

so yeah, the feature extraction is not working at all. This is because of the geometry of your acquisition, which is composed of many short but dense ligns. Is this aerial LiDAR?

I see a few things that could improve the results, with minimal coding. the first one would be to increase the pruning to mitigate the varying density. You mention that you use  `--voxel_width 50` but when I run this I obtain a cloud of 31 points (wich make sense as your point cloud is 250 * 250 unit wide). A more reasonable value might probably be 0.5 (which gives 341 562 points).

Do check what is going on when you run line 214 of partition.py and paste the terminal log, which should look somewhat like that:
```=========================
======== pruning ========
=========================
Voxelization into 501 x 537 x 39 grid
Reduced from 8924118 to 341562 points (3.82%)```

If the pruning doesn't work I have other ideas, but they will be more involved.

moemen95, 2019-03-12:
> 178832

hello how much it took to generate sp graph with your 9 million points ?!

loicland, 2019-03-13:
Hi,

You should either:
- prune your input point cloud so it is smaller
- use an external library for the delaunay triangulation. Check out [this issue](https://github.com/loicland/superpoint_graph/issues/86) for suggestions.

moemen95, 2019-03-16:
Thank you so much. i reduced it with voxel width thanks.

closed 2019-04-23 by loicland