loicland/superpoint_graph #89: read_las returning XYZ as raw integers, instead of actual coordinates

ghgr, 2019-02-24:
In *partition/provider.py*, Line [286-288] the function *read_las* looks as follows (note the uppercase X,Y and Z):

```python  N_points = len(inFile)
    x = np.reshape(inFile.X, (N_points,1))
    y = np.reshape(inFile.Y, (N_points,1))
    z = np.reshape(inFile.Z, (N_points,1))
```

In *laspy* there is a difference between uppercase and lowercase coordinates. According to [the docs](https://laspy.readthedocs.io/en/latest/tut_part_1.html):
> Upper case dimensions (las_file.X, las_file.Y, las_file.Z) give the raw integer dimensions, while lower case dimensions (las_file.x, las_file.y, las_file.z) give the scaled value. 

Since the *read_las* method is supposed to return **xyz** (and optionally extra data), shouldn't these lines be changed to lowercase? Like:

```python  N_points = len(inFile)
    x = np.reshape(inFile.x, (N_points,1))
    y = np.reshape(inFile.y, (N_points,1))
    z = np.reshape(inFile.z, (N_points,1))
```

loicland, 2019-02-25:
I personally don't use .las, but it seems that you are correct. Have you been successfully using SPG with the lower case dimensions?

ghgr, 2019-02-25:
Yes, in fact, it only works when using the lower case dimensions. The uppercase dimensions are usuarlly large values (in the order of 1e7), which when computing volumes we get values in the order of ~1e21. Then, when *learning/spg.py* calculates 
```python 
node_att['volume'] = np.maximum(0, f['sp_volume'][:] ** 2)
```
we would get something in the ~1e42 order. Since that overflows the range of float32, it is set as *+Inf". 

This *+Inf* will then raise an Exception when calling *StandardScaler*.

At first I bypassed this error by forcing *node_att['volume']* as *np.float64*, until I figured out why the coordinates were so large.

loicland, 2019-02-25:
Good call then. Pushing the change right now. Thanks!

closed 2019-04-23 by loicland