loicland/superpoint_graph #90: Issues on the S3DIS Dataset when testing

Yajha, 2019-03-03:
Hi.

When i run test code on the S3DIS Dataset (Partition Part is OK) , the program will always stuck on this situtaion(82%), Hou could i fix this?

[image]

Thank you for your work!

loicland, 2019-03-04:
Hi,

Does it always crash at the same time? On the 8th run of he inference on the first fold at 82%? What about the other folds?

Are you monitoring your GPU memory? Could it be that it saturates?

I think this could be related to [https://github.com/loicland/superpoint_graph/issues/81](this issue). Maybe try the "attempt" fix.

Let me know if it doesn't fix it.

Yajha, 2019-03-05:
Thank you very much, the code works with `--test_multisamp_n 1`

To your question,
Yes, it always gets stuck on the 8th run on the first fold at 82% with no error information.I have check the GPU memory and there is no problem.
But I still don't know why the problem occurs.

Thanks for your time!

Yajha, 2019-03-05:
Thank you very much.
The code works with --test_multisamp_n 1.But I still don't know why that problem occurs.

Thanks for your time.

| |
袁嘉
|
|
yj_seu2013@163.com
|
签名由网易邮箱大师定制

closed 2019-03-06 by loicland