loicland/superpoint_graph #93: how to improve the performance of trained spg model?

cbyzju, 2019-03-18:
Dear Loic:
      I trained and applied spg model to my own dataset, everything is fine, except that the performance of trained model is not very good, shown as the attachment.  In this figure,  car is red, tree is green, and the building is blue, my trained spg model can segment them from the pointcloud correctly, while the performance for the brown wall is very bad,  many points within the wall are segmented as car, tree or building, and this phenomenon is common for the whole map,  I wonder how to solve this problem, and get better result?  Thank you very much!

[image]

loicland, 2019-03-18:
Hi,

a few questions:
- how large is your training set
- what performance do you reach on the training set
- is there at least one similar brown wall in your training set and if yes is it classified correctly?
- are you using the normalized xyz features? if yes, are they consistent from one point cloud to another?

cbyzju, 2019-03-19:
hi loic:

1. My training set has around 1 billion points,  the training set consists of several blocks of city A, several blocks of city Z, and so on.
2. I applied the trained model to my training set,  the phenomenon I described above is still exist, object A are segmented as the mixture of object A, B, C and so on, not only the brown wall, but also for the other objects.
3.  With respect to normalized xyz features,  I consider you mean "spg.scaler01" normalization, is it correct? I use the same parameters for training and test dataset.

cbyzju, 2019-03-19:
hi loic:
I also have another question. Since I want to improve the performance of trained model, I add more training data into the training process gradually,  but there is no any help for the improvement of the performance. Do you have any suggestions?

loicland, 2019-03-19:
Hi,

1 - ok this should be more than enough. A common source of error is the computation of the elevation (line 83 of `learning/s3dis_dataset.py` for example. Make sure that it is consistent from one point cloud to another.
2 - Does the partition work ok? Can you post the partition of the scene above (`output_type --p` in `partition/visualize.py`)?
It seems like your acquisition is a little bit irregular. You could try to subsample your point cloud to regularize the density (maybe using `--voxel_width 0.05` for example), with added benefit that the partition step will be much faster.
Can you post a view of geometric features of the scene above as well (`output_type --f` in `partition/visualize.py`)? if they are too messy , you could be to remove the geometric features from the pointnets' input: `--pc_attribs xyzrgbe` and 
`--ptn_nfeat_stn 7`

3 - no, scaler is for the superedge features. I am referring to `--pc_attribs`. make sure that you don't have a similar line to line 88 of `learning/s3dis_dataset.py`. What is the size of the pointnet's input? You can see that in the printed model at the beginning of the learning phase.
4 - I am not sure what you mean by 
>add more training data into the training process gradually
You should run the model for a few hundred epochs on your entire traing set at once, and validate the progress on an unseen validation set every 10 or so epochs when prototyping.

Let me know if it helps! It looks like an exciting dataset.

cbyzju, 2019-03-20:
hi loic,
1. For my training and test dataset, I use the format of semantic3d, and use ` elpsv[:,0] /= 100  elpsv[:,1:] -= 0.5` for all the point cloud. And these is no a similar line to line 88 of `learning/s3dis_dataset.py`, do you recommend using normalized xyz features?
2. For the purpose of faster computation time and lower overfitting,  I use `--voxel_width 0.05`for downsampling,  and `--reg_strength 0.15` for minimal partition.
3. With respect to the performance of partition and extracted geometric features,  I select four typical errors and you can check it. ( top left is geometric features, top right is prediction results, partition results is below)
[image]
[image]
For these two images, red cars are not segmented correctly.
[image]
[image]
For these two images, pink curbs are also wrong.

loicland, 2019-03-21:
Something is going wrong clearly.

- can you make sure that the value of z is consistent over your dataset? As of now, SPG assumes that the ground is perfectly flat, but if yopu have hills and slopes it can mess up the values of the elevation. There is a simple enough fix to that, I will try to incorporate it soon.
- could you give me a small sample of your dataset so I can see for my self? Like a google drive link, just a few hundred thousand points would be enough. Parttiion, prediction and ground truth would be useful.
- you can try to increase the voxel_size of the pruning to decrease the geometric noise which seems to be rather high.
- i assume you don't have radiometric information. Can you post the result of `print(model)` so I can make sure it was correctly taken into account?
- I absolutely do not encourage to use scaled position if you are not dealing with indoor data.
- for the curbs the problem is that they are not properly segmented. What would be the ground truth here? just the edge of the curb or the entire sidewalk? If the former you need to decrease the regularization strength.
- Finally your problem seems to be a good candidate for learned superpoints. We will release an improves segmentation step this April, I will let you know when. Maybe this will help.
- Are you showing me images from the training or testing set?

closed 2019-04-23 by loicland