loicland/superpoint_graph #94: confused with label=0 in、

justbin1123, 2019-03-18:


loicland, 2019-03-18:
see this issues:

https://github.com/loicland/superpoint_graph/issues/83
https://github.com/loicland/superpoint_graph/issues/63
https://github.com/loicland/superpoint_graph/issues/45

let me know if this doesn't clear up your confusion.

justbin1123, 2019-03-18:
> _No description provided._

loicland, 2019-03-18:
Hi,

label 0 is reserved in semantic3d for unannotated data, which are excluded from the evaluation metric. We have followed the same standard. Hence 0 should only be reserved for unannotated data. If you don't have any, start the labels at 1.

However, since our newtork associates a label to all (super)points, the output starts at 0, which corresponds to the label 1 in the input. A little bit confusing I know; but it made the evaluation functions much easier for us.

Let me know if this doesn't completely clear things up for you.

justbin1123, 2019-03-19:
Thanks a lot！
“label 0 is reserved in semantic3d for unannotated data, which are excluded from the evaluation metric. ” Is that mean label 0  isn’t involved in training（/back propagation ）？how did u do it？

loicland, 2019-03-19:
> mean label 0 isn’t involved in training

correct. We are sometimes dealing with point clouds which are sparsely annotated, and don't want unlabelled point to influennce the training phase. Note that this is different from a 'background' class, as the label unnannotated is never assigned by the network.

The label of superpoints is the  majority label of the points it contains. In the event where a superpoint contains no labeled point, it is assigned the label -100.
See line 69 of `training/spg.py` and line 375 of `main.py`.

justbin1123, 2019-03-19:
get it！ 
Thank u so much！

closed 2019-03-19 by justbin1123