loicland/superpoint_graph #98: AttributeError: module 'torch.nn._functions.thnn' has no attribute 'rnnFusedPointwise'

josch, 2019-03-21:
Hi,

after following the instructions in `README.md` and running

    python3 learning/main.py --dataset s3dis --S3DIS_PATH ../S3disData/ --cvfold 1 --epochs 350 --lr_steps '[275,320]' --test_nth_epoch 50 --model_config 'gru_10_0,f_13' --ptn_nfeat_stn 14 --nworkers 2 --odir "results/s3dis/best/cv1"

I get the following output:

```
Will save to results/s3dis/best/cv1
/home/user/superpoint_graph/learning/graphnet.py:28: UserWarning: nn.init.orthogonal is now deprecated in favor of nn.init.orthogonal_.
  if orthoinit: init.orthogonal(fnet_modules[-1].weight, gain=init.calculate_gain('relu'))
/home/user/superpoint_graph/learning/graphnet.py:32: UserWarning: nn.init.orthogonal is now deprecated in favor of nn.init.orthogonal_.
  if orthoinit: init.orthogonal(fnet_modules[-1].weight)
/home/user/superpoint_graph/learning/pointnet.py:42: UserWarning: nn.init.constant is now deprecated in favor of nn.init.constant_.
  nn.init.constant(self.proj.weight, 0); nn.init.constant(self.proj.bias, 0)
Total number of parameters: 279409
Module(
  (ecc): GraphNetwork(
    (0): RNNGraphConvModule(
      (_cell): GRUCellEx(
        32, 32
        (ini): InstanceNorm1d(1, eps=1e-05, momentum=0.1, affine=False, track_running_stats=False)
        (inh): InstanceNorm1d(1, eps=1e-05, momentum=0.1, affine=False, track_running_stats=False)
        (ig): Linear(in_features=32, out_features=32, bias=True)
      )(ingate layernorm)
      (_fnet): Sequential(
        (0): Linear(in_features=13, out_features=32, bias=True)
        (1): ReLU(inplace)
        (2): Linear(in_features=32, out_features=128, bias=True)
        (3): ReLU(inplace)
        (4): Linear(in_features=128, out_features=64, bias=True)
        (5): BatchNorm1d(64, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
        (6): ReLU(inplace)
        (7): Linear(in_features=64, out_features=1024, bias=False)
      )
    )
    (1): Linear(in_features=352, out_features=13, bias=True)
  )
  (ptn): PointNet(
    (stn): STNkD(
      (convs): Sequential(
        (0): Conv1d(14, 64, kernel_size=(1,), stride=(1,))
        (1): BatchNorm1d(64, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
        (2): ReLU(inplace)
        (3): Conv1d(64, 64, kernel_size=(1,), stride=(1,))
        (4): BatchNorm1d(64, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
        (5): ReLU(inplace)
        (6): Conv1d(64, 128, kernel_size=(1,), stride=(1,))
        (7): BatchNorm1d(128, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
        (8): ReLU(inplace)
      )
      (fcs): Sequential(
        (0): Linear(in_features=128, out_features=128, bias=True)
        (1): BatchNorm1d(128, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
        (2): ReLU(inplace)
        (3): Linear(in_features=128, out_features=64, bias=True)
        (4): BatchNorm1d(64, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
        (5): ReLU(inplace)
      )
      (proj): Linear(in_features=64, out_features=4, bias=True)
    )
    (convs): Sequential(
      (0): Conv1d(14, 64, kernel_size=(1,), stride=(1,))
      (1): BatchNorm1d(64, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
      (2): ReLU(inplace)
      (3): Conv1d(64, 64, kernel_size=(1,), stride=(1,))
      (4): BatchNorm1d(64, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
      (5): ReLU(inplace)
      (6): Conv1d(64, 128, kernel_size=(1,), stride=(1,))
      (7): BatchNorm1d(128, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
      (8): ReLU(inplace)
      (9): Conv1d(128, 128, kernel_size=(1,), stride=(1,))
      (10): BatchNorm1d(128, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
      (11): ReLU(inplace)
      (12): Conv1d(128, 256, kernel_size=(1,), stride=(1,))
      (13): BatchNorm1d(256, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
      (14): ReLU(inplace)
    )
    (fcs): Sequential(
      (0): Linear(in_features=257, out_features=256, bias=True)
      (1): BatchNorm1d(256, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
      (2): ReLU(inplace)
      (3): Linear(in_features=256, out_features=64, bias=True)
      (4): BatchNorm1d(64, eps=1e-05, momentum=0.1, affine=True, track_running_stats=True)
      (5): ReLU(inplace)
      (6): Linear(in_features=64, out_features=32, bias=True)
    )
  )
)
Epoch 0/350 (results/s3dis/best/cv1):
  0%|                                                                       | 0/114 [00:00<?, ?it/s]/home/user/superpoint_graph/learning/pointnet.py:131: UserWarning: volatile was removed and now has no effect. Use `with torch.no_grad():` instead.
  out = model.ptn(Variable(clouds, volatile=True), Variable(clouds_global, volatile=True))
THCudaCheck FAIL file=/pytorch/aten/src/THC/THCGeneral.cpp line=405 error=11 : invalid argument
/home/user/.local/lib/python3.7/site-packages/torch/nn/functional.py:1332: UserWarning: nn.functional.sigmoid is deprecated. Use torch.sigmoid instead.
  warnings.warn("nn.functional.sigmoid is deprecated. Use torch.sigmoid instead.")
Traceback (most recent call last):
  File "learning/main.py", line 387, in <module>
    main()
  File "learning/main.py", line 286, in main
    acc, loss, oacc, avg_iou = train()
  File "learning/main.py", line 184, in train
    outputs = model.ecc(embeddings)
  File "/home/user/.local/lib/python3.7/site-packages/torch/nn/modules/module.py", line 489, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/user/superpoint_graph/learning/graphnet.py", line 97, in forward
    input = module(input)
  File "/home/user/.local/lib/python3.7/site-packages/torch/nn/modules/module.py", line 489, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/user/superpoint_graph/learning/modules.py", line 58, in forward
    hx = self._cell(input, hx)
  File "/home/user/.local/lib/python3.7/site-packages/torch/nn/modules/module.py", line 489, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/user/superpoint_graph/learning/modules.py", line 112, in forward
    state = torch.nn._functions.thnn.rnnFusedPointwise.GRUFused
AttributeError: module 'torch.nn._functions.thnn' has no attribute 'rnnFusedPointwise'
```

josch, 2019-03-21:
Downgrading torch to version 0.4.1.post2 fixed this problem.

Please adapt superpoint_graph so that it works with the latest version of torch.

Thanks!

loicland, 2019-04-10:
Hi,

for some reason `torch.nn._functions.thnn.rnnFusedPointwise.GRUFused` does not exist anymore in 1.0.

A quick fix would simply be to change line 108 of `learning/modules.py`
from:
```
if input.is_cuda:
```
to
```
if input.is_cuda and torch.__version__.split('.')[0]=='0' :
```
Looking into a better fix for next update.

mys007, 2019-04-24:
This is probably related to https://github.com/pytorch/pytorch/pull/10305 , so my suggestion would be:
```
cuda_fused_gru_cell = torch._C._VariableFunctions._thnn_fused_gru_cell
return cuda_fused_lstm_cell(gi, gh, hidden, self.bias_ih, self.bias_hh)
```
I haven't tried it myself, though.

closed 2019-12-23 by loicland