loicland/superpoint_graph #103: Issue with number of classes in metrics.py file

Amrsaeed, 2019-03-23:
I am trying to test the trained network on the full Semantic3D dataset using the command 

`CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset sema3d --SEMA3D_PATH $SEMA3D_DIR --db_test_name testfull --db_train_name trainval \
--epochs -1 --lr_steps '[350, 400, 450]' --test_nth_epoch 100 --model_config 'gru_10,f_8' --ptn_nfeat_stn 11 \
--nworkers 2 --odir "results/sema3d/trainval_best" --resume RESUME`

And I am receiving an invalid value encountered in divide error resulting from this function

`  def get_average_intersection_union(self):
    values = self.get_intersection_union_per_class()
    class_seen = ((self.confusion_matrix.sum(1)+self.confusion_matrix.sum(0))!=0).sum()
    return sum(values) / class_seen`

Apparently class_seen is 0
Any idea how to fix?

loicland, 2019-03-25:
Hi,

A few questions: 
- are you running the code with any modification ?
- does the bug happen at the very first iteration or later?
- can you print out `print(confusion_matrix.confusion_matrix)` line 205 of `learning/main.py`?

Amrsaeed, 2019-03-25:
Hi,

- There are no modifications on the code
-  This happens during the final evaluation
-  The confusion matrix is zero. I have been tracing it and it seems that this happens only when using test set as tvec_cpu is zero since there are no ground truths for the test cases. So 
```
count_predicted_batch
```
 function gets zeros for the ground truths. Is that normal behaviour?

loicland, 2019-03-25:
Ah I see. indeed, as there are no ground truth in the test files, the confusion matrix is empty. So this is normal, there is no way to compute metrics on unlabelled data.

Does this error causes a crash for you? Or just an alert? Are the results files written correctly?

Amrsaeed, 2019-03-25:
It results in a RuntimeWarning for the zero division. I think it might cause a crash on other versions. Maybe add a check for the ground truth of the test set to avoid the warning and misleading results.

closed 2019-04-23 by loicland