loicland/superpoint_graph #106: error in running partition.py

amardeepjaiman, 2019-04-02:
I am getting following error while running partition.py for my custom Data (all files in .ply format).

    creating the feature file...
Dataset/MyDataset/data/train/000027.ply
=========================
======== pruning ========
=========================
Voxelization into 2000 x 120 x 1 grid
Reduced from 135640 to 4641 points (3.42%)
=========================
======== pruning ========
=========================
Voxelization into 2000 x 130 x 1 grid
Reduced from 4641 to 4618 points (99.5%)

    computing the superpoint graph...
        minimal partition...
L0-CUT PURSUIT WITH L2 FIDELITY
PARAMETERIZATION = FAST
Graph 4620 vertices and 110832 edges and observation of dimension 4
Iteration   1 -    3 components - Saturation   0.0 % - Quadratic Energy 96.601 % - Timer  0.283808
Iteration   2 -    3 components - Saturation   7.1 % - Quadratic Energy 96.601 % - Timer  0.526575
Stopping criterion reached
        computation of the SPG...
Traceback (most recent call last):
  File "partition/partition.py", line 184, in <module>
    graph_sp = compute_sp_graph(xyz, args.d_se_max, in_component, components, labels, n_labels)
  File "/data/home/amjn_cowi/amardeep_AI_work/Pointcloud_classification/superpoint_graph/partition/graphs.py", line 56, in compute_sp_graph
    tri = Delaunay(xyz)
  File "qhull.pyx", line 1828, in scipy.spatial.qhull.Delaunay.__init__
  File "qhull.pyx", line 354, in scipy.spatial.qhull._Qhull.__init__
scipy.spatial.qhull.QhullError: QH6154 Qhull precision error: Initial simplex is flat (facet 1 is coplanar with the interior point)

While executing:  | qhull d Qbb Qc Qt Q12 Qz
Options selected for Qhull 2015.2.r 2016/01/18:
  run-id 1624726386  delaunay  Qbbound-last  Qcoplanar-keep  Qtriangulate
  Q12-no-wide-dup  Qz-infinity-point  _pre-merge  _zero-centrum  Qinterior-keep
  Pgood  _max-width 1e+03  Error-roundoff 6.9e-09  _one-merge 6.2e-08
  Visible-distance 4.1e-08  U-coplanar-distance 4.1e-08  Width-outside 8.3e-08
  _wide-facet 2.5e-07

precision problems (corrected unless 'Q0' or an error)
      2 flipped facets
   4617 zero divisors during gaussian elimination

The input to qhull appears to be less than 4 dimensional, or a
computation has overflowed.

Qhull could not construct a clearly convex simplex from points:
- p1702(v4): 6e+05 5.7e+06  -0.5  0.22
- p387(v3): 6e+05 5.7e+06  -0.5  0.54
- p4479(v2): 6e+05 5.7e+06  -0.5 1.8e-12
- p0(v1): 6e+05 5.7e+06  -0.5  0.42
- p4607(v0): 6e+05 5.7e+06  -0.5  0.16

The center point is coplanar with a facet, or a vertex is coplanar
with a neighboring facet.  The maximum round off error for
computing distances is 6.9e-09.  The center point, facets and distances
to the center point are as follows:

center point 5.955e+05 5.679e+06     -0.5   0.2683

facet p387 p4479 p0 p4607 distance=    0
facet p1702 p4479 p0 p4607 distance=    0
facet p1702 p387 p0 p4607 distance=    0
facet p1702 p387 p4479 p4607 distance=    0
facet p1702 p387 p4479 p0 distance=    0

These points either have a maximum or minimum x-coordinate, or
they maximize the determinant for k coordinates.  Trial points
are first selected from points that maximize a coordinate.

The min and max coordinates for each dimension are:
  0:  5.95e+05  5.96e+05  difference= 1000
  1:  5.679e+06  5.679e+06  difference=   65
  2:      -0.5  -2.225e-308  difference=  0.5
  3:  1.819e-12      1000  difference= 1000

If the input should be full dimensional, you have several options that
may determine an initial simplex:
  - use 'QJ'  to joggle the input and make it full dimensional
  - use 'QbB' to scale the points to the unit cube
  - use 'QR0' to randomly rotate the input for different maximum points
  - use 'Qs'  to search all points for the initial simplex
  - use 'En'  to specify a maximum roundoff error less than 6.9e-09.
  - trace execution with 'T3' to see the determinant for each point.

If the input is lower dimensional:
  - use 'QJ' to joggle the input and make it full dimensional
  - use 'Qbk:0Bk:0' to delete coordinate k from the input.  You should
    pick the coordinate with the least range.  The hull will have the
    correct topology.
  - determine the flat containing the points, rotate the points
    into a coordinate plane, and delete the other coordinates.
  - add one or more points to make the input full dimensional.

Please note, i get this error in one of the ply file partitioning, some other ply files got processed successfully. Please suggest what can be done.

loicland, 2019-04-02:
Hi,

It seems that your point cloud is quite small. In particular, the partition step only finds 3 superpoints. You should probably use a lower regularization strength, unless they are very simple clouds (in which case, you may not need to use SPG and can use directly pointnet maybe).

Now onto the probelm with Delaunay. The python function is not able to compute a meaningful partition, in particular it thinks your data is on a flat plane. One thing that I can think of is that you have a huge offset, you probably use absolute coordinates. This is a problem as we use float32 to encode position. If your point cloud is too small in extent compared to its position it is going to cause underflow errors.

One simple thing that you could do would be to shift your point clouds such that they are centered in 0 directly in your `read_custom_dataset` function (using a float64 to read them).

Let me know if this fix your problem or if this is unclear to you.

amardeepjaiman, 2019-04-03:
Hi @loicland ,

Thanks for prompt response but i am afraid i am still unclear about it. could you please explain below :
1. what do you mean by simple clouds here ? 
2. Let me tell you more about my dataset. I am having a large dense aerial point cloud for around 50 sq km area. Data is classified in 2 classes , ground and non-ground (any point other then ground like buildings, water, any other structure etc). I have split the whole point cloud into multiple point cloud tiles(~30)  probably 5sq km each.

could you tell me if SPG approach is suitable for this kind of dataset ? If you like I can share sample data with you. so that you can better suggest.

3. The problem i stated above occured with 1 point cloud file not all. So i have excluded that from the dataset. Do you suggest to go ahead with training ?

Thanks in advance

loicland, 2019-04-04:
Hi,

1- your data is very flat. Indeed the voxels grid used for pruning has only one voxel width on the z axis. Worst, they seem to all have the exact same value. Do the shift I mentioned such that the xyz values are centered around 0 and it should help. Unless your cloud 000027.ply is really extremely flat.

2 - You should use SPG if: youa re convinced that considering the entire point cloud at once is important as you have a global structure that is determining. This might not be the case for a simple groun-nonground classification. To be honnest you could probably use pointnet with a sliding window or pointnet++ with subsampling and obtain better results. Unless speed and scaling is a major issue for you.
For SPG it is important to have a meaningful elevation value. In your case, it might not be necessary as ground-nonground is so easy. So either build a very simple elevation model (with RANSAC to find a plane for example), or discard the elevation alltogether if it is not consistent from one cloud to another.
3 - I would not go on for the training without fixing the shift issue, and visualizing the ground truth and partition files to, insure that they make sense (you can post images here if you want my opinion).

closed 2019-04-23 by loicland